tiny-pilot/tinypilot #389: PS/2 keyboard adapter

kesp91, 2020-12-22:
Hello, I am running an old server repurposed to unraid (Supermicro MB X8HAD) and tiny pilot and see video output but keyboard does not work. There are ps/2 connectors and was wondering if would be possible to utilize them/has anyone had a solution for this.

mtlynch, 2020-12-23:
Another project [found a way to do it through PS/2](https://zach.bloomqu.ist/blog/2020/08/pilo-raspberry-pi-lights-out-management.html#power-control-via-ps2), but TinyPilot's probably not going to go that route.

I'm experimenting with a new keyboard USB descriptor. Can you try running this command and testing to see if it's compatible with your motherboard?

```bash
export TINYPILOT_INSTALL_VARS="tinypilot_repo_branch=new-kb-hid"
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

closed 2020-12-31 by mtlynch

mtlynch, 2021-04-08:
@shawnallen85 - No keyboard or mouse functionality at all? 

The issue is that the Pi has only one USB OTG port, to it presents itself to the target system as a USB hub with a USB keyboard and USB mouse attached. A lot of KVMs expect a direct connection to the USB keyboard/mouse, so they ignore input from USB hubs. If you try connecting a USB keyboard to the KVM via a USB hub and it still doesn't work, there's likely nothing TinyPilot can do to achieve compatibility.

I'm maintaining a list of KVMs and their compatibility with TinyPilot here:

https://github.com/mtlynch/tinypilot/wiki/KVM-compatibility