tiny-pilot/tinypilot #396: Support keyboard-only mode

mtlynch, 2020-12-31:
TinyPilot registers itself as a multifunction USB device, but some BIOS software fails to recognize multifunction USB.

To improve compatibility with those devices, TinyPilot settings should allow the user to switch to "keyboard only mode" which would change the [TinyPilot USB gadget](https://github.com/mtlynch/tinypilot/blob/33d3cd5630ea17c4ce02eb436f0cdf436d72ac0a/scripts/usb-gadget/init-usb-gadget) to impersonate a simple USB keyboard instead.

ComputerDoktor, 2021-01-05:
might be a solution for the problem mentioned in https://github.com/mtlynch/tinypilot/issues/372

mtlynch, 2021-01-26:
I'm experimenting with this in this branch:

https://github.com/mtlynch/ansible-role-tinypilot/commits/keyboard-only

WarheadsSE, 2021-02-15:
@mtlynch I experienced this "kind" of problem with a KVM I have, which has a dedicated keyboard input port, which is USB 1.1 compliant only. In order to make use of the KVM, I simply needed to skip linking the mouse functions dir into the configs dir. I could get a mouse to work from TinyPilot, but only one device descriptor at a time.

My "problem" seems to be in the report & messaging by TinyPilot itself, as I have been able to make use of several multi-function USB devices (Logitech Unifiy Receiver, Acer/Lite-On kb/mouse) without issues on the same port.