tiny-pilot/tinypilot #397: Mouse forwarding fails when connecting to a Raspberry Pi target

mtlynch, 2021-01-06:
## Description

A user reported finding buggy mouse behavior when they connected their TinyPilot to a Raspberry Pi with `graphical.target`  started running LXDE.

## What's the behavior that you expect?

Mouse cursor forwarding works as normal.

## What's happening instead?

Mouse movements and clicks are substantially delayed on the remote Pi.

## What are the steps to reproduce this behavior?

1. Provision a target Pi with LXDE
1. Connect to the Pi from a TinyPilot

ComputerDoktor, 2021-04-22:
https://github.com/pikvm/pikvm/blob/master/pages/faq.md#usb-problems-keyboard-mouse-mass-storage-etc
Unusual case: RPi4 is used as a Pi-KVM to control RPi3. In this case, the mouse delay may be several seconds.
To fix it, just add kernel parameter usbhid.mousepoll=0 in /boot/cmdline.txt on the server (i.e. RPI3 in our case) and reboot it.
(Parameter has to be added to cmdline.txt, not to config.txt)
Raspi OS kernel has a nonsense default value of 0xffffffff for usbhid.mousepoll. 
more details here https://wiki.archlinux.org/index.php/mouse_polling_rate

mtlynch, 2021-04-28:
Oh, thanks for sharing that solution, @ComputerDoktor!

closed 2021-04-28 by mtlynch