tiny-pilot/tinypilot #403: Docker support

tristan-k, 2021-01-13:
Is it possible to install tinypilot with docker and is or will there be a docker available at dockerhub?

mtlynch, 2021-01-13:
I love Docker, so I wish it could, but unfortunately this isn't possible. TinyPilot requires changes to the Pi's `/boot/config.txt` and `/etc/modules`, so it's not possible to encapsulate that in a Docker container.

closed 2021-01-13 by mtlynch

LalitMaganti, 2021-01-13:
Looking at the ansible role [1], apart from the installation of the USB gadget, I think pretty much everything else could run in a Docker container. Adding Docker support would add a lot of value to aid repeatability, cross platform compatibility and allow easily put a forward proxy (e.g. Traefik) in-front of tinypilot.

Is there something I'm missing in my analysis?

I'm waiting for my USB capture device to arrive to test out tinypilot but I may take a crack at implementing this if this is something which would be accepted upstream :)

[1] https://github.com/mtlynch/ansible-role-tinypilot/blob/master/tasks/main.yml

mtlynch, 2021-01-13:
I'm open to the Docker container, but I'd like to understand more about what the goal is. Without the USB gadget, there's no keyboard or mouse input, so what would a Dockerized TinyPilot do?

LalitMaganti, 2021-01-13:
The idea is that the docker container runs all the layers above the USB gadget (i.e. ngnix, ustreamer, tinypilot) with the USB gadget needing to be installed separately by itself.

There's precedent for this in other projects: for example, the Wireguard docker container [1]. This uses the kernel module which is expected to be installed on the bare-metal machine with all the userspace code running in the container. (tbh with this example there's also a mode which actually compiles *and* runs the module inside the container itself but I don't think that's possible in the TinyPilot case).

[1] https://hub.docker.com/r/linuxserver/wireguard

mtlynch, 2021-01-13:
Unfortunately, I don't think I'd accept that upstream.

I love Docker, but I can't think of a way to justify it. I wouldn't make it the default way to run in production because it adds too much complexity. And normally I'd love Docker as a way to test in CI, but in this case, I think it would mean that my CI build would be too different from my production build.

I appreciate the proposal and sharing details from Wireguard. I'm not totally closed to the idea long term, but I just don't see a way that it fits into the software without becoming essentially an extra compatibility target to maintain. Ansible is kind of wonky, but it's at least the devil that I know at this point.

LalitMaganti, 2021-01-13:
That's fair I totally understand. I'll maintain something in my personal collection of dockerfiles in that case :)

mtlynch, 2021-01-13:
Let me know if you end up creating it. I'd love to take a look and see if I can find a way to integrate it.

lamarios, 2021-01-15:
I've done it:

Ustreamer  Dockerifle
```
FROM ubuntu

RUN apt update && apt -y install git build-essential libevent-dev libjpeg62-dev uuid-dev libbsd-dev make gcc libjpeg8 libjpeg-turbo8 libuuid1 libbsd0

RUN git clone --depth=1 https://github.com/pikvm/ustreamer

WORKDIR /ustreamer

RUN make && make install && ustreamer --help

EXPOSE 8080

CMD ["ustreamer", "--host=0.0.0.0" ,"-r", "1920x1080" ,"--persistent" ,"--drop-same-frames=30", "-l"]
```

(somehow the compilation stopped working today. I'll need to investigate)

TinyPilot dockerfile
```
FROM python:3

RUN apt-get update

RUN curl -sL https://deb.nodesource.com/setup_14.x |  bash -
RUN  apt-get install -y nodejs

RUN git clone https://github.com/mtlynch/tinypilot

WORKDIR /tinypilot

RUN pip install --requirement requirements.txt && pip install --requirement dev_requirements.txt
RUN npm install prettier@2.0.5

RUN ./dev-scripts/build

EXPOSE 8000
ENV PORT=8000

CMD ["python", "/tinypilot/app/main.py"]
```

Nginx  template file
kvn.conf.template
```
server {
	listen 10080 ;
	server_name your.domain.com;

	location = /stream {
		proxy_pass http://ustreamer:8080/stream;
		proxy_send_timeout 1200s;
		proxy_read_timeout 1200s;
		proxy_set_header Upgrade $http_upgrade;
		proxy_set_header X-Forwarded-Proto $scheme;
	        proxy_set_header Connection "Upgrade";
		fastcgi_send_timeout 1200s;
		fastcgi_read_timeout 1200s;
		proxy_buffering off;
		proxy_set_header X-Real-IP $remote_addr;
		proxy_set_header Host            $host;
		proxy_set_header X-Forwarded-Host  $host;
		add_header Strict-Transport-Security "max-age=15552000; includeSubDomains; preload";
    	}

	location / {
		proxy_pass http://tinypilot:8000;
		proxy_send_timeout 1200s;
		proxy_read_timeout 1200s;
		proxy_set_header Upgrade $http_upgrade;
		proxy_set_header X-Forwarded-Proto $scheme;
	        proxy_set_header Connection "Upgrade";
		fastcgi_send_timeout 1200s;
		fastcgi_read_timeout 1200s;
		proxy_buffering off;
		proxy_set_header X-Real-IP $remote_addr;
		proxy_set_header Host            $host;
		proxy_set_header X-Forwarded-Host  $host;
		add_header Strict-Transport-Security "max-age=15552000; includeSubDomains; preload";
	}
}
```

docker-compose file
```yml
version: '3.6'
services:
  nginx:
    image: nginx
    container_name: nginx
    restart: always
    depends_on:
     - ustreamer
     - tinypilot
    ports:
     - 10080:10080
    volumes:
     - ./nginx:/etc/nginx/templates
  ustreamer:
    build:
      context: ./images/ustreamer/
    container_name: ustreamer
    restart: always
    ports:
      - "8181:8080"
    devices:
      - /dev/video0:/dev/video0
  tinypilot:
    build:
      context: ./images/tinypilot/
    container_name: tinypilot
    restart: always
    ports:
      - "8000:8000"
    devices:
    - /dev/hidg0:/dev/hidg0
    - /dev/hidg1:/dev/hidg1

```

There's probably a way to skip on the  nginx config files by using traeffik and set up reverse proxying with  labels, I'll need to look into it.

mtlynch, 2021-01-15:
This is really cool! Thanks for sharing, @lamarios!

What other configuration was needed outside of docker-compose? You need to configure the USB gadgets separately, right?

lamarios, 2021-01-16:
Yeah that's correct. By the way, is there a way to undo the usb script ? Now I always reboot when there's an update but there's probably a better way.

lamarios, 2021-01-16:
Ok updated with traefik for more simplicity and remove the need for extra local files.

```yml
version: '3.6'
services:
  traefik:
    restart: always
    image: traefik:2.3
    command:
      - "--log.level=DEBUG"
      - "--api.insecure=true"
      - "--providers.docker=true"
      - "--providers.docker.exposedbydefault=false"
      - "--entrypoints.web.address=:80"
    container_name: traefik
    ports:
      - "10080:80"
      - "18080:8080"
    volumes:
      - "/var/run/docker.sock:/var/run/docker.sock:ro"
  ustreamer:
    build:
      context: ./images/ustreamer/
    container_name: ustreamer
    restart: always
    labels:
    - "traefik.enable=true"
    - "traefik.http.routers.ustreamer.rule=Host(`yourdomain.com`) && PathPrefix(`/stream`)"
    - "traefik.http.services.ustreamer.loadbalancer.server.port=8080"
    devices:
      - /dev/video0:/dev/video0
  tinypilot:
    build:
      context: ./images/tinypilot/
    container_name: tinypilot
    labels:
    - "traefik.enable=true"
    - "traefik.http.routers.kvm.rule=Host(`yourdomain.com`)"
    - "traefik.http.services.kvm.loadbalancer.server.port=8000"
    restart: always
    devices:
    - /dev/hidg0:/dev/hidg0
    - /dev/hidg1:/dev/hidg1
```

mtlynch, 2021-01-18:
Nice, thanks for sharing this!

>By the way, is there a way to undo the usb script ?

I believe there is, but I haven't experimented with this much. I started looking into this recently while investigating the best way to implement resetting virtual storage.

I think I accidentally reset everything by either writing an empty string to `/sys/kernel/config/usb_gadget/g1/UDC` or by removing it, but I haven't had an opportunity to tinker with it further.

If you end up finding an answer, I'd be interested.

sobake, 2021-03-25:
> Ok updated with traefik for more simplicity and remove the need for extra local files.
> 
> Edit: I checked for ustreamer and they actually have docker files in their repo but it doesn't seem like they have official images yet so i built and published one
> 
> ```yaml
> version: '3.6'
> services:
>   traefik:
>     restart: always
>     image: traefik:2.3
>     command:
>       - "--log.level=DEBUG"
>       - "--api.insecure=true"
>       - "--providers.docker=true"
>       - "--providers.docker.exposedbydefault=false"
>       - "--entrypoints.web.address=:80"
>     container_name: traefik
>     ports:
>       - "10080:80"
>       - "18080:8080"
>     volumes:
>       - "/var/run/docker.sock:/var/run/docker.sock:ro"
>   ustreamer:
>     image: gonzague/ustreamer
>     container_name: ustreamer
>     restart: always
>     command: ["ustreamer", "--host=0.0.0.0" ,"-r", "1920x1080" ,"--persistent" ,"--drop-same-frames=30", "-l"]
>     labels:
>     - "traefik.enable=true"
>     - "traefik.http.routers.ustreamer.rule=Host(`yourdomain.com`) && PathPrefix(`/stream`)"
>     - "traefik.http.services.ustreamer.loadbalancer.server.port=8080"
>     devices:
>       - /dev/video0:/dev/video0
>   tinypilot:
>     build:
>       context: ./images/tinypilot/
>     container_name: tinypilot
>     labels:
>     - "traefik.enable=true"
>     - "traefik.http.routers.kvm.rule=Host(`yourdomain.com`)"
>     - "traefik.http.services.kvm.loadbalancer.server.port=8000"
>     restart: always
>     devices:
>     - /dev/hidg0:/dev/hidg0
>     - /dev/hidg1:/dev/hidg1
> ```

What else is needed to run tinypilot via docker? I can't seem to find the usb script that is referred to... Thanks!

lamarios, 2021-03-26:
The usb script is the only thing that you need to run manually.

tristan-k, 2021-08-07:
@lamarios Can you got in to more detail how to set this up with portainer? I'm trying to run it on a x86 machine (Intel Atom x5-Z8350) and for now it would be enough to get TinyPilots video streaming working. Copying and pasting your docker-compose in to portainer results in an error.

[image]

lamarios, 2021-08-08:
I don't use portainer but the error message is clear enough. You're trying to build the image from a path that doesn't exist.