tiny-pilot/tinypilot #406: Shift lost

oliverkrystal, 2021-01-15:
I'm running TinyPilot on a Pi Zero with an HDMI to CSI camera interface.  In both Firefox and Chromium web browser, when I attempt to capitalize a letter tiny pilot sends the shift keystroke, then the capital letter.

However, what happens on the remote device the letter is not capitalized.

If I press and hold shift, then 0, release the shift and type l, then the screen displays "ol".
If I press and hold shift, then o, then l, then release shift "oL" is typed.
If I hit caps lock and type o, then I get "O"

Is this just a result of the lower processing power of the pi zero?

mtlynch, 2021-01-15:
Thanks for reporting this! I don't think the Zero W is going to give you a great experience because it's super slow, but I can try to figure out what's going wrong.

Can you share a screenshot of what you're seeing in the web interface? Make sure that the keystroke history is turned on.

Can you also share your logs by running this command?

```bash
/opt/tinypilot/dev-scripts/dump-logs
```

oliverkrystal, 2021-01-16:
You're welcome.  Right now, being slow will be okay.  I mostly put this together for experimentation and the random odd job.

Below is the log and screenshots connected to the freenas machine and then using my primary laptop to connect to tiny pilot.

```
TinyPilot log dump
https://tinypilotkvm.com
Timestamp: 2021-01-16T03:45:51+00:00

Software versions
TinyPilot version: 1.3.0-5-gbf07bfe bf07bfe
uStreamer version: v1.17-85-g28563ab 28563ab
OS version: Linux raspberrypi 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux

voltage logs

TinyPilot configuration
[Unit]
Description=TinyPilot - RPi-based virtual KVM
After=syslog.target network.target

[Service]
Type=simple
User=tinypilot
WorkingDirectory=/opt/tinypilot
ExecStart=/opt/tinypilot/venv/bin/python app/main.py
Environment=HOST=127.0.0.1
Environment=PORT=8000
Environment=KEYBOARD_PATH=/dev/hidg0
Environment=MOUSE_PATH=/dev/hidg1
Restart=always

[Install]
WantedBy=multi-user.target

TinyPilot logs
-- Logs begin at Fri 2021-01-15 06:02:16 GMT, end at Sat 2021-01-16 03:45:53 GMT. --
Jan 15 06:02:44 raspberrypi systemd[1]: Started TinyPilot - RPi-based virtual KVM.
Jan 15 06:02:57 raspberrypi python[468]: [2021-01-15 06:02:57,500] INFO in main: Starting app
Jan 15 06:02:57 raspberrypi python[468]: [2021-01-15 06:02:57,906] INFO in _internal:  * Restarting with stat
Jan 16 03:44:56 raspberrypi python[468]: [2021-01-16 03:44:56,630] INFO in main: Starting app
Jan 16 03:45:00 raspberrypi python[468]: [2021-01-16 03:45:00,792] INFO in socket_api: Client connected
Jan 16 03:45:03 raspberrypi python[468]: [2021-01-16 03:45:03,217] INFO in socket_api: Client connected

uStreamer configuration
[Unit]
Description=uStreamer - Lightweight, optimized video encoder
After=syslog.target network.target

[Service]
Type=simple
User=ustreamer
WorkingDirectory=/opt/ustreamer
ExecStart=/opt/ustreamer/ustreamer \
  --host 127.0.0.1 \
  --port 8001 \
  --encoder omx \
  --format uyvy \
  --workers 3 \
  --drop-same-frames 30 \
  --persistent \
  --dv-timings \

Restart=always

[Install]
WantedBy=multi-user.target

uStreamer logs
-- Logs begin at Fri 2021-01-15 06:02:16 GMT, end at Sat 2021-01-16 03:45:53 GMT. --
[... 15 lines cut ...]
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.687    stream] -- Using pixelformat: UYVY
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.687    stream] -- Querying HW FPS changing is not supported
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.688    stream] -- Using IO method: MMAP
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.706    stream] -- Requested 2 device buffers, got 3
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.723    stream] -- Capturing started
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.723    stream] -- Initializing BCM ...
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.765    stream] -- Initializing OMX ...
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.775    stream] -- Initializing OMX encoder ...
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.820    stream] -- Initializing OMX encoder ...
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [36.865    stream] -- Initializing OMX encoder ...
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [37.045    stream] -- Using JPEG quality: 80%
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [37.047    stream] -- Creating pool with 3 workers ...
Jan 15 06:02:45 raspberrypi ustreamer[466]: -- INFO  [37.049    stream] -- Capturing ...
Jan 16 03:45:02 raspberrypi ustreamer[466]: -- INFO  [72.468      http] -- HTTP: Registered client: [127.0.0.1]:38728, id=3e23d2f8-4afa-4fd9-b190-75599656939b; clients now: 1

nginx logs
-- Logs begin at Fri 2021-01-15 06:02:16 GMT, end at Sat 2021-01-16 03:45:53 GMT. --
Jan 15 06:02:45 raspberrypi systemd[1]: Starting A high performance web server and a reverse proxy server...
Jan 15 06:02:45 raspberrypi systemd[1]: Started A high performance web server and a reverse proxy server.

2021/01/16 03:44:48 [error] 486#486: *1 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.1.120, server: tinypilot, request: "GET /socket.io/?EIO=3&transport=polling&t=NS9IwLF HTTP/1.1", upstream: "http://127.0.0.1:8000/socket.io/?EIO=3&transport=polling&t=NS9IwLF", host: "192.168.1.142", referrer: "http://192.168.1.142/"
2021/01/16 03:44:54 [error] 486#486: *1 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.1.120, server: tinypilot, request: "GET /socket.io/?EIO=3&transport=polling&t=NS9IxqY HTTP/1.1", upstream: "http://127.0.0.1:8000/socket.io/?EIO=3&transport=polling&t=NS9IxqY", host: "192.168.1.142", referrer: "http://192.168.1.142/"
```
[image]
[image]

oliverkrystal, 2021-01-16:
For grins, I connected this to my primary laptop and then fired up an older laptop (it has VGA and a CD-ROM drive oh my).

There, it worked as expected.
[image]

oliverkrystal, 2021-01-17:
So, I also attached this to a Pi4 (as a retropie box which will probably get more use).

The shift/capilatization works as expected.

I will have to see what's up with the NAS?

mtlynch, 2021-01-18:
Oh! That's strange. Does the NAS show expected output when you plug in a regular keyboard?

oliverkrystal, 2021-01-20:
So, rebooted just to be sure and tinyPilot still fails to send the shift/alt/etc.

However, an external USB keyboard reacts as expected.

[image]
[image]

mtlynch, 2021-01-21:
That's really strange. While connected to the NAS, can you try typing these commands on the Pi OS shell?

```bash
echo -ne "\x20\0\xb\0\0\0\0\0" > /dev/hidg0 && \
  echo -ne "\0\0\xc\0\0\0\0\0" > /dev/hidg0 && \
  echo -ne "\0\0\0\0\0\0\0\0" > /dev/hidg0
```

That should produce the text `Hi`.

oliverkrystal, 2021-01-22:
That works as expected, it types Hi.

[image]

mtlynch, 2021-01-22:
Hmm, that's strange. That suggests that it's a problem with the frontend (i.e., the JavaScript code).

At first, I was thinking it might be that the Pi Zero W is slow, so it lets the keystroke requests get queued and then processes them not necessarily in the order they arrived, but that wouldn't explain why you see different behavior on different target systems.

We're sending keystroke messages over websockets, and I thought websockets guaranteed the ordering, but I'm not totally sure about that.

It could also be that there's a race condition on sending the keystroke to the target. TinyPilot [spawns separate processes](https://github.com/mtlynch/tinypilot/blob/6447d7623615dc2d044eb725b252d8a954b777ce/app/hid/write.py) to do the writes, but it's possible that there's enough overhead in creating a process that process B gets to the write before process A and they're out of order, but again, it doesn't explain the different results on different target systems.

>Of note, if I use cord to split out the 5v power with a y cable, typing doesn't work at all on this system where the same cable did work with a pi4.

Heads up about Y-cables, unless they have protections in place (I've never seen one that does), there's [a risk of reverse current](https://github.com/mtlynch/tinypilot/wiki/Powering-your-TinyPilot-safely) if you connect it to the Pi and an independently powered computer.

oliverkrystal, 2021-01-22:
I'm wondering if its something odd in how FreeNAS does the terminal.
Sometimes a message gets displayed and the TinyPilot keystrokes get ignored.

I really need to upgrade the system to the latest and I wonder if that
wouldn't fix the issue (or more correctly make it a nonissue).  I guess
I'll have to try to accomplish that this weekend.

I built the y-cable and dropped the powerline from the host usb to the pi.

oliverkrystal, 2021-07-17:
I'm going to go ahead and close this issue.  After all the futzing around I did ..... I do not recommend running tinypilot on a Pi Zero W.  It can/may work.  HDMI will usually work, keyboard probably works.  But I continually ran into trouble and things not quite right.

After getting ahold of a Pi3 for another project HDMI worked reliably, every time.  Upgrading to the Pi4 (that other project is on going) I've had USB keyboard and HDMI working reliable.  The USB3 ports on the system even provide enough power for the Pi4 to run without a splitter.

If all you've got is a pi zero then you can try it but my recommendation straight up is to get a Pi4 on order.

closed 2021-07-17 by oliverkrystal