tiny-pilot/tinypilot #409: customizable appearance to distinguish.

yamato225, 2021-01-18:
I use multiple TinyPilot to control multiple machines, I sometimes confuse which machine do I control. 
I want to change appearance of the web interface to distinguish.
For example:

* add color customizable frame around the monitor
* add header and show text(e.g. host name or set by user)

These appearances are disabled as default.

How do you feel?

mtlynch, 2021-01-18:
Oh, that's interesting. More users are using multiple TinyPilots in their environment, and it would be good if there's a way to distinguish them.

How do you envision the UI for setting these options?

lensherm, 2021-03-12:
@mtlynch, Question along these lines, but not quite: of course I could use a KVM switch feeding into TinyPilot, but that naturally allows only one remote machine to be visible and controlled by the end user at any given time from one instance of TinyPilot. Are there any theoretical resource (CPU/RAM/network) bottlenecks you would envision if two or more HDMI capture devices were all feeding an rpi 4 running TinyPilot? On a large-enough monitor, I could envision maybe a 4x4 grid of remote screens, with USB input being switched to, let's say, whichever machine has the mouse pointer over it's screen, at any given time.