tiny-pilot/tinypilot #410: Power cycling a system with GPIO pins

rubyrose96, 2021-01-19:
Started playing with tintpilot the other day and loving it so far. 
Having the ability to emulate the power button would be useful, seeing as the pi has GPIO headers it'd be easy to hook it up to a [relay module](https://thepihut.com/products/adafruit-stemma-non-latching-mini-relay) to be able to turn on or force shudown a system.
How does this sound?

mtlynch, 2021-01-19:
Thanks for reaching out!

I'm definitely interested in power management, but I want to keep the solution something that an average IT pro could set up. All the relay module options I've seen involve soldering or setting up breadboards and plugging directly into the motherboard, and that's very different from the rest of the way TinyPilot works.

I'm looking into a packaged offering based on [this device](https://dlidirect.com/products/iot-power-relay), but I need to do some work there because out of the box, that power strip still requires jumper wire.

closed 2021-01-19 by mtlynch

rubyrose96, 2021-01-21:
That's fair enough, I see how it'd be the easier option for an end used to implement. 
Do you have any pointers as to how something like my idea could be implemented? I'm a still new to python so my knowledge is a bit lacking, as this is open source I'd like to play around with it a bit.

mtlynch, 2021-01-21:
You might want to check out this project, as it's similar but it implemented power management:

https://github.com/Fmstrat/diy-ipmi