tiny-pilot/tinypilot #433: Change device hostname through web UI

mtlynch, 2021-01-22:
As more users are deploying multiple TinyPilots to the same network, we should make it easier to change their hostnames.

Which changes the hostname to the desired

### Ansible

Add a script to `/opt/tinypilot-privileged` that the `tinypilot` user can execute as sudo to change the hostname in `/etc/hosts` and `/etc/hostname`.

### Backend

Add a new REST endpoint:

* `/api/hostname` (POST)

The API should accept a JSON message like this:

```javascript
{
  "hostname": "some-new-name"
}
```

* `/api/hostname` (GET)

Retrieves the current hostname like this:

```javascript
{
  "success": true,
  "error": null,
  "hostname": "tinypilot"
}
```

### Frontend

We should add an option under "Commands" that's "Change Hostname". It pops up a dialog box that says "Change hostname" and offers a text input showing the current hostname (from `local_system.hostname`) and allows the user to change it.

If the user changes it, the dialog should say: "Restart now to apply changes?"

If the user chooses to reboot, the dialog should call `/api/shutdown` to reboot and then poll in the background for a response at the new hostname URL. When the new URL is serving, the dialog should redirect the browser to the new URL.

jotaen, 2021-02-12:
@mtlynch on a general note, do you have something in mind for how to go about the local dev server? Because right now it’s not mandatory to start the dev server in an environment where the priviledged scripts are available under `/opts/...`, so using these settings via the UI locally will produce errors. Along with the “shutdown” commands this is an additional case where the [dev and prod environments are not at par](https://12factor.net/dev-prod-parity).

mtlynch, 2021-02-12:
>on a general note, do you have something in mind for how to go about the local dev server?

I don't have a good plan for that yet. It started out with the shutdown as the only privileged command, but then I've been slowly adding new ones over the past month and so there's more and more drift.

One thing we could do is define a configuration for an Ansible installation you can run in a dev environment so that it's closer to the prod setup and has the right scripts in place.

jotaen, 2021-02-12:
How about mocking the scripts? (Or was that what you meant?) It’s not exactly parity, on the other hand there is little point in, e.g., changing the network config such as the host name, or shutting down the machine, when it’s all running locally. For me it would already be a good first step if the frontend didn’t crash in an odd way.

mtlynch, 2021-02-12:
I was thinking of putting the real scripts in, but we could mock as well. I started a separate bug in #501 to track.

jotaen, 2021-02-24:
I believe this is done, isn’t it? 🥳

mtlynch, 2021-02-24:
Oh right. Fixed!

closed 2021-02-24 by mtlynch

AphinityAvali, 2025-04-25:
Multiple devices I've tried on the public fork of tinypilot will not accept anything entered in the hostname field, giving a warning of what characters are permitted each time.

db39, 2025-04-28:
Hi @AphinityAvali, I'm sorry you ran into this error when setting a hostname using `.local` as a suffix. 

>it does appear to infer .local after reboot, so this is more of a UI confirmation that you do not need the .local

That's correct - you don't need to specify, `.local` here. I can see how it can be confusing to have the error appearing to say the format is invalid when you want to use `.local`.

[image]

I've created a [new GitHub issue](https://github.com/tiny-pilot/tinypilot/issues/1879) to edit the error message to mention not needing `.local` (or similar). Please let me know if you have any questions or suggestions on this.