tiny-pilot/tinypilot #437: Cannot find hidg0 & hidg1 on RPI4 with latest raspbian

darkr4y, 2021-01-23:
## Description

```
pi@raspberrypi4b:~ $ uname -a
Linux raspberrypi4b 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

pi@raspberrypi4b:~ $ sudo ls -al /dev/h*
crw------- 1 root root 244,   0 Jan 23 15:44 /dev/hidraw0
crw------- 1 root root  10, 183 Jan 23 15:44 /dev/hwrng
```
Cannot find hidg0 & hidg1 on RPI4 with latest raspbian, and tinypilot show the following error:

```
....
Jan 23 15:48:16 raspberrypi4b python[536]: Process Process-3:
Jan 23 15:48:16 raspberrypi4b python[536]: Traceback (most recent call last):
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 23 15:48:16 raspberrypi4b python[536]:     self.run()
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jan 23 15:48:16 raspberrypi4b python[536]:     self._target(*self._args, **self._kwargs)
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/opt/tinypilot/app/hid/write.py", line 17, in _write_to_hid_interface_immediately
Jan 23 15:48:16 raspberrypi4b python[536]:     with open(hid_path, 'wb+') as hid_handle:
Jan 23 15:48:16 raspberrypi4b python[536]: PermissionError: [Errno 13] Permission denied: '/dev/hidg1'
Jan 23 15:48:16 raspberrypi4b python[536]: Process Process-4:
Jan 23 15:48:16 raspberrypi4b python[536]: Traceback (most recent call last):
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 23 15:48:16 raspberrypi4b python[536]:     self.run()
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jan 23 15:48:16 raspberrypi4b python[536]:     self._target(*self._args, **self._kwargs)
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/opt/tinypilot/app/hid/write.py", line 17, in _write_to_hid_interface_immediately
Jan 23 15:48:16 raspberrypi4b python[536]:     with open(hid_path, 'wb+') as hid_handle:
Jan 23 15:48:16 raspberrypi4b python[536]: PermissionError: [Errno 13] Permission denied: '/dev/hidg1'
Jan 23 15:48:16 raspberrypi4b python[536]: Process Process-5:
Jan 23 15:48:16 raspberrypi4b python[536]: Traceback (most recent call last):
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 23 15:48:16 raspberrypi4b python[536]:     self.run()
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jan 23 15:48:16 raspberrypi4b python[536]:     self._target(*self._args, **self._kwargs)
Jan 23 15:48:16 raspberrypi4b python[536]:   File "/opt/tinypilot/app/hid/write.py", line 17, in _write_to_hid_interface_immediately
Jan 23 15:48:16 raspberrypi4b python[536]:     with open(hid_path, 'wb+') as hid_handle:
Jan 23 15:48:16 raspberrypi4b python[536]: PermissionError: [Errno 13] Permission denied: '/dev/hidg1'
Jan 23 15:48:17 raspberrypi4b python[536]: Process Process-6:
Jan 23 15:48:17 raspberrypi4b python[536]: Traceback (most recent call last):
Jan 23 15:48:17 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 23 15:48:17 raspberrypi4b python[536]:     self.run()
Jan 23 15:48:17 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jan 23 15:48:17 raspberrypi4b python[536]:     self._target(*self._args, **self._kwargs)
Jan 23 15:48:17 raspberrypi4b python[536]:   File "/opt/tinypilot/app/hid/write.py", line 17, in _write_to_hid_interface_immediately
Jan 23 15:48:17 raspberrypi4b python[536]:     with open(hid_path, 'wb+') as hid_handle:
Jan 23 15:48:17 raspberrypi4b python[536]: PermissionError: [Errno 13] Permission denied: '/dev/hidg0'
Jan 23 15:48:17 raspberrypi4b python[536]: Process Process-7:
Jan 23 15:48:17 raspberrypi4b python[536]: Traceback (most recent call last):
Jan 23 15:48:17 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 23 15:48:17 raspberrypi4b python[536]:     self.run()
Jan 23 15:48:17 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jan 23 15:48:17 raspberrypi4b python[536]:     self._target(*self._args, **self._kwargs)
Jan 23 15:48:17 raspberrypi4b python[536]:   File "/opt/tinypilot/app/hid/write.py", line 17, in _write_to_hid_interface_immediately
Jan 23 15:48:17 raspberrypi4b python[536]:     with open(hid_path, 'wb+') as hid_handle:
Jan 23 15:48:17 raspberrypi4b python[536]: PermissionError: [Errno 13] Permission denied: '/dev/hidg0'
Jan 23 15:48:18 raspberrypi4b python[536]: Process Process-8:
Jan 23 15:48:18 raspberrypi4b python[536]: Traceback (most recent call last):
Jan 23 15:48:18 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jan 23 15:48:18 raspberrypi4b python[536]:     self.run()
Jan 23 15:48:18 raspberrypi4b python[536]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jan 23 15:48:18 raspberrypi4b python[536]:     self._target(*self._args, **self._kwargs)
Jan 23 15:48:18 raspberrypi4b python[536]:   File "/opt/tinypilot/app/hid/write.py", line 17, in _write_to_hid_interface_immediately
Jan 23 15:48:18 raspberrypi4b python[536]:     with open(hid_path, 'wb+') as hid_handle:
Jan 23 15:48:18 raspberrypi4b python[536]: PermissionError: [Errno 13] Permission denied: '/dev/hidg0'
....
```
## What's the behavior that you expect?

Works as intended

## What's happening instead?

Not works as intended

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Enter a few characters at random
4. Use `sudo journalctl -u tinypilot` to see error

## Screenshots

*If applicable, add screenshots to help explain your problem.*

## Logs

https://paste.ubuntu.com/p/TMv8p4NCsC/

mtlynch, 2021-01-25:
Thanks for reporting this!

It looks like the USB gadget initializer never ran. Did you reboot after installing TinyPilot?

Can you share the output from this command?

```bash
sudo journalctl -u usb-gadget
```

darkr4y, 2021-01-26:
Yes, I rebooted several times
here is the usb gadget logs:

```
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + cp /tmp/tmp.YgqPTatYJe functions/hid.keyboard/report_desc
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + MOUSE_FUNCTIONS_DIR=functions/hid.mouse
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + mkdir -p functions/hid.mouse
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo 0
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo 0
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo 5
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: ++ mktemp
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + D=/tmp/tmp.WrELQCgErH
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x05\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x09\x02'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\xA1\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x05\x09'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x19\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x29\x08'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x15\x00'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x25\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x95\x08'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x75\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x81\x02'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x05\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x09\x30'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x09\x31'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x16\x00\x00'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x26\xFF\x7F'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x75\x10'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x95\x02'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x81\x02'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x09\x38'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x15\x81'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x25\x7F'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x75\x08'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x95\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x81\x06'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x05\x0C'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x0A\x38\x02'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x15\x81'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x25\x7F'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x75\x08'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x95\x01'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\x81\x06'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo -ne '\xC0'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + cp /tmp/tmp.WrELQCgErH functions/hid.mouse/report_desc
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + CONFIG_INDEX=1
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + CONFIGS_DIR=configs/c.1
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + mkdir -p configs/c.1
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo 250
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + CONFIGS_STRINGS_DIR=configs/c.1/strings/0x409
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + mkdir -p configs/c.1/strings/0x409
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + echo 'Config 1: ECM network'
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + ln -s functions/hid.keyboard configs/c.1/
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + ln -s functions/hid.mouse configs/c.1/
Jan 26 00:00:46 raspberrypi4b init-usb-gadget[395]: + ls /sys/class/udc
Jan 26 00:00:47 raspberrypi4b init-usb-gadget[395]: + chmod 777 /dev/hidg0
Jan 26 00:00:47 raspberrypi4b init-usb-gadget[395]: chmod: cannot access '/dev/hidg0': No such file or directory
Jan 26 00:00:47 raspberrypi4b systemd[1]: usb-gadget.service: Main process exited, code=exited, status=1/FAILURE
Jan 26 00:00:47 raspberrypi4b systemd[1]: usb-gadget.service: Failed with result 'exit-code'.
Jan 26 00:00:47 raspberrypi4b systemd[1]: Failed to start Initialize USB gadgets.
```

It still looks like the hidg0 device is not found.

darkr4y, 2021-01-27:
I think I know the cause of the problem
follow the script https://github.com/mtlynch/key-mime-pi/blob/master/enable-usb-hid
I checked the `/boot/config.txt` and found thant there was no `dtoverlay=dwc2` in it, add this line to the file, and now tinypilot works fine.
its time to close this issue,but wait ...
I'm still curious as to why the tinypilot quick install script doesn't add this action ?

closed 2021-01-27 by darkr4y