tiny-pilot/tinypilot #438: RPi Zero W ustreamer Requested 2 device buffers, got 3  ERROR Unable to start capturing: Invalid argument

terxw, 2021-01-23:
## Description

I have two problems:

1. cannot connect to the web interface
2. cannot get ustreamer working

Hardware:

rpi zero w Lusya HDMI to CSI-2 Adapter Board

software:

raspberry os lite

install tinypilo kvm with:

export TINYPILOT_INSTALL_VARS="ustreamer_capture_device=tc358743 tinypilot_interface=0.0.0.0"
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot

## What's the behavior that you expect?

web interface accesible
stream working

## What's happening instead?

web browser is ubable to connect

## Logs

http://sprunge.us/psoRPp

mtlynch, 2021-01-25:
Glad to hear you got this figured out!

mtlynch, 2021-01-25:
Also, `tinypilot_interface` is for the the backend. Nginx listens to 0.0.0.0 on port 80 and forwards the requests to the TinyPilot backend at localhost:8000.

There generally isn't a reason to have the TinyPilot backend also listen on 0.0.0.0.

closed 2021-01-25 by mtlynch