tiny-pilot/tinypilot #439: Add ShellCheck to Circle CI config

mtlynch, 2021-01-25:
I just discovered [shellcheck](https://github.com/koalaman/shellcheck).

It would be a great addition to this project so that we have an automated way of enforcing best practices on shell code and ensuring consistent style conventions.

We should add this to the [Circle CI config](https://github.com/mtlynch/tinypilot/blob/037234b4dd6e46e84a6ad4c10ef6d5555dc5b479/.circleci/config.yml) using the [docker image](https://hub.docker.com/r/koalaman/shellcheck). I know there's a CircleCI orb, but I prefer having the CircleCI config be transparent and easy for developers to replicate on their local development machine.

ossiemoore, 2021-02-15:
Not sure this is helpful, so feel free to discard. I am going through an online self study thing for Python and the task you described felt like a good first 'real world' task for me to write in Python. I know shell scripting too, so if you *prefer* a shell script, I would be happy to oblige. Just let me know. 

Save the attached `code.txt` as `go.py` (or whatever you want to call it) and make it executable.
> Note that the code has a Linux/macOS `#!/usr/bin/python3` directive at line 1. If on Windows, replace `./go.sh` below with `python go.sh`.

Scan the current directory. Expects `shellcheck` to be in your path.
```
./go.py
```
Scan `/tmp` directory. Expects `shellcheck` to be in your path.
```
./go.py /tmp
```
Scan `/tmp` directory using `shellcheck` located in `/usr/bin/shellcheck`.
```
./go.py /tmp /usr/bin/shellcheck
```

Source:
[code.txt](https://github.com/mtlynch/tinypilot/files/5979041/code.txt)

ossiemoore, 2021-02-15:
FYI.. if you really just need a `find` command to do the same thing, this would find all `*.sh` files in your home `bin` directory, run `shecllcheck` against each one and log all results into `sc.log`

```
find ~/bin -type f -name "*.sh" -exec shellcheck -o all {} \;  > sc.log
```

ossiemoore, 2021-02-15:
Here is another example.. (1) find all files and (2) grep non-binary hits for literal string `#!/bin/sh`, printing the filename only (3) pipe that to awk to craft the shellcheck statement to parse the file and finally (4) pipe that to `/bin/sh` to execute the line.

```
find ~/bin -type f -exec grep -I -l '#!/bin/sh' {} \; | awk '{print "shellcheck "$0}'  | sh
```

mtlynch, 2021-02-15:
@ossiemoore thanks for working on this!

I'd want it to be implemented in bash because I think it's a task better suited to bash. Your last example looks pretty good, but a few notes:

* We want to run `find` on the source repository rather than the `~/bin` directory
* The shebang line we use is `#!/bin/bash`
* Our shebang line is always the first line in a file, so it might be easier to run `head -n 1` and compare it to the expected value rather than searching for the line anywhere in the file.
* Instead of calling `awk` to create a shell command, I think it's cleaner to do something like this:

```bash
readonly BASH_FILES="$(find ...)"
shellcheck "${BASH_FILES}"
```

closed 2021-03-12 by jdeanwallace