tiny-pilot/tinypilot #448: Change uStreamer settings from the web UI

mtlynch, 2021-01-28:
I need to think through the implementation more, but it would be very helpful if users could adjust uStreamer settings directly from the web UI.

Currently, they have to edit their `/lib/systemd/system/ustreamer.service` file by hand, reload, and restart, which is cumbersome.

We should offer simple ways to adjust resolution, frame rate, and video quality.

mtlynch, 2021-04-22:
Let's hold off on adding resolution settings to the UI. I've been playing around with it on the Voyager, and I can't get uStreamer to accept any other resolution than the default, so it'll be a confusing user experience if they try changing the resolution and nothing happens.

Changing JPEG quality is still useful, though.

jdeanwallace, 2021-04-22:
Okay cool. I know you're testing on the Voyager, but I didn't have any problems changing the uStreamer resolution using the HDMI-to-USB dongle. [Video demo](https://www.loom.com/share/37ea82ea0a814cc993da4349497392d7).

mtlynch, 2021-04-22:
Yeah, it's successful for me on the USB dongle as well, but I can't figure out how to get it to change on the Voyager. I'm not sure if the TC358743 chip just doesn't support it and uStreamer won't resize down or if I'm just missing something.

PR #667 (2021-04-23): Change uStreamer jpeg quality from the web UI
This PR adds "JPEG Quality" to the `video-settings-dialog` custom element. [Video demo](https://www.loom.com/share/8b5dc0a54c0746b19ee6ebcea11c4cf8).

This is a continuation of the work done in #629 of allowing video settings to be changed via the web UI.

This relies on https://github.com/mtlynch/ansible-role-ustreamer/pull/39 to be merged.

Fixes #448

---
This change is [[image]](https://reviewable.io/reviews/mtlynch/tinypilot/667)
closed 2021-04-23 by jdeanwallace