tiny-pilot/tinypilot #449: Loss of video on reboot - keyboard still works

matthewdavis, 2021-01-28:
## Description

On a reboot, video is lost but the keyboard still works. Simply unplugging and replugging the HDMI cable into the adapter fixes and video is restored.

## What's the behavior that you expect?

Video to work during and after a reboot.

## What's happening instead?

No Video.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Verify video is working
4. Login and issue a reboot command

## Logs

http://sprunge.us/OhW4uI

matthewdavis, 2021-01-28:
So I am using the adapter data/power 'tinypilot' adapter provided. As well as the usb power brick provided. I have not swapped any cables or anything away from what was provided in the hobbyist kit. I also tried plugging the USB dongle into the other USB-3 port with the same results.

Btw, a reboot of the tinypilot will not fix the issue. The only way I've been able to fix it is to re-plug the HDMI cable.

mtlynch, 2021-01-29:
Hmm, that shouldn't be happening. It sounds like an issue with either the HDMI dongle or the power adapter. I'll ship you free replacement parts. I'll email you first to confirm the same shipping address.

Sorry for the inconvenience!

closed 2021-01-29 by mtlynch

matthewdavis, 2021-01-29:
I know you've closed it and issued replacement parts and closed this case. Thank-you. I'm putting some logs here for posterity sake. I ran another test and rebooted the machine. It actually worked over 1 reboot, and I saw the BIOS messages. Then on the 2nd reboot, it went dead and changed from a 'grey' color to a solid black screen. I could type in it, but no video.

> Jan 29 18:58:17 tinypilot python[513]: [2021-01-29 18:58:17,892] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
> Jan 29 18:58:20 tinypilot python[513]: [2021-01-29 18:58:20,322] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
> Jan 29 18:59:10 tinypilot python[513]: [2021-01-29 18:59:10,191] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
> Jan 29 18:59:10 tinypilot python[513]: [2021-01-29 18:59:10,255] ERROR in write: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
> Jan 29 18:59:10 tinypilot python[513]: Process Process-63:
> Jan 29 18:59:10 tinypilot python[513]: Traceback (most recent call last):
> Jan 29 18:59:10 tinypilot python[513]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap 
> Jan 29 18:59:10 tinypilot python[513]:     self.run()
> Jan 29 18:59:10 tinypilot python[513]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
> Jan 29 18:59:10 tinypilot python[513]:     self._target(*self._args, **self._kwargs)
> Jan 29 18:59:10 tinypilot python[513]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
> Jan 29 18:59:10 tinypilot python[513]:     hid_handle.write(bytearray(buffer))
> Jan 29 18:59:10 tinypilot python[513]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
> Jan 29 18:59:35 tinypilot python[513]: [2021-01-29 18:59:35,129] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
> Jan 29 18:59:35 tinypilot python[513]: [2021-01-29 18:59:35,656] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?