tiny-pilot/tinypilot #465: Extremely high latency when multiple client windows are open

WardBenjamin, 2021-02-03:
## Description

If you open the frontend web interface multiple times (for the sameTinyPilot), latency increases to 5-6 seconds or worse, including sometimes totally freezing up.

## What's the behavior that you expect?

If TinyPilot doesn't function with multiple streams at the same time, I expect the focused window to take over all input. Ideally the other display would still show the display output.

## What's happening instead?

Latency increases massively and the software becomes unusable.

## What are the steps to reproduce this behavior?

1. Open the TinyPilot web frontend in multiple tabs or windows or browsers.

## Screenshots

N/A

## Logs

http://sprunge.us/ynhMsj

WardBenjamin, 2021-02-04:
I've had a chance to test a bit further, and I didn't see the same problem if I ran the stream on multiple computers (instead of multiple streams on the same computer).

mtlynch, 2021-02-04:
Thanks for reporting this!

I suspect that this is a consequence of MJPEG. It's an uncommon format, and it consumes a lot of bandwidth. So the latency could be because you're flooding your network links with multiple MJPEG streams. Or, given that it works fine from multiple computers, it could be that multiple streams are overwhelming the browser.

We're planning to move to H264 in the next few months, likely with WebRTC, so I think this issue will naturally disappear as a consequence, but I'll keep it open to track.

mtlynch, 2023-02-16:
This is now fixed with H.264 streaming mode.

closed 2023-02-16 by mtlynch