tiny-pilot/tinypilot #466: Add Mechanism to Abort Paste

ossiemoore, 2021-02-04:
Once a large paste begins, you cannot abort the keystrokes gracefully. Provide a graceful way to stop the paste. In a clipboard tool I wrote for myself, I do this by stopping playback of the clipboard contents if you wiggle your mouse.

mtlynch, 2021-02-04:
Thanks for reporting this!

Can you tell me more about the scenario? Pastes generally complete so fast that there wouldn't be a way to give the user a sensible way to cancel it.

I'm wondering if the actual issue is pasting something you didn't mean to. In which case, would it be helpful to have a dialog beforehand like, "You're about to paste 800 characters. Are you sure you want to continue?"

ossiemoore, 2021-02-06:
For me, when I copy text from my local machine and then paste that text into the remote machine, the "paste" operation executes slowly. Because it pastes slowly, if something pops up and alert/etc., the remaining text begins to "type" into that popup window. Fortunately, the work around is to close the browser window, which aborts the typing.

I attached a short 42 second video showing the issue.

https://user-images.githubusercontent.com/44754148/107108096-18797f80-67fb-11eb-87f1-30fbf99a8da1.mp4

FYI.. you'll see a note on the screen about Pastey, my clipboard management tool. I was going to demonstrate how I handle this problem in it but decided not to bloat this GitHub issue with that video since, for you, the solution is simpler and more straight forward. (This isn't a  plug for it as it is a personal development tool that I don't publish or make available publicly.)

jotaen, 2021-02-24:
On a slightly related note, wouldn’t it be more transparent UX-wise if the paste overlay was a textarea with two buttons “paste” and “cancel”? So you would paste into the textarea first, which allows you to make adjustments to the copy and revisit what you are actually about to paste. Sometimes there is unintended whitespace in the paste copy, or a character is missing because you didn’t select the original text carefully enough.

ossiemoore, 2021-02-24:
FYI.. I like the suggestion from @jotaen. That would be a great addition/alternative.

mtlynch, 2021-02-24:
My hesitation with the paste confirmation UX is that it's different than almost all other apps. For example, when you paste text into a terminal window, it doesn't ask you to paste in a temporary buffer first and then confirm the paste. It assumes that the user intended to paste whatever's in their clipboard.

The overlay is just a hacky workaround for the fact that browsers don't allow apps to access the clipboard over HTTP. Over HTTPS connections, I think we should be able to just skip the overlay portion entirely and paste directly as soon as the user clicks "Command > Paste."