tiny-pilot/tinypilot #469: Quirky Login Process for Tunneled Connections

ossiemoore, 2021-02-04:
## Description

I use an SSH tunnel when connecting to TinyPilot from out in the wild. This works perfectly fine, but the login sequence is a bit quirky.

## What's the behavior that you expect?

Hit URL for my Tiny Pilot and be presented with a login screen.

## What's happening instead?

After hitting the initial URL, I get a could not connect because it redirects to http. I manually update the URL again and all is well.

## What are the steps to reproduce this behavior?

1. SSH into a Linux server on the same network as your TinyPilot, tunneling some port to the TinyPilot's IP address and 443 SSL port.
2. Connect to https://localhost[:<port>]/
3. Browser will redirect to http://localhost/login
4. Update the protocol to https and add the port you specified in step and press enter.
5. Login and do your thing.

## Screenshots

not necessary

## Logs

http://sprunge.us/lsojCr

mtlynch, 2021-02-04:
Thanks for reporting this!

Can you share the SSH command you're using (feel free to replace any hostnames with `example`)?

ossiemoore, 2021-02-04:
You bet, here you go:
```
ssh <user>@<remote_server> -L <localport>:<hostname/ip_from_remote_server_perspective>:<port_tinypilot_is_listening_on>
```
For example, assume you are on your laptop at a hotel. You have a Linux server at home that you can SSH into. Assume its' public IP address is `123.123.123.123`. Assume that your TinyPilot IP at home has an ip of `192.168.1.205` and that it is listing on the default port of `443`. You decide you are going to tunnel local port `64443`.

1. SSH to your Linux server at home, tunneling your laptop `local` port 64443 to your TinyPilot (192.168.2.205) port 443 on the `remote` (home) network:
```
ssh john.smith@123.123.123.123 -L 64443:192.168.1.205:443
```
2. Now you can open the web browser on your laptop and browse to `http://localhost:64443` to access your TinyPilot all while not exposing your TinyPilot to the public internet.

ossiemoore, 2021-02-04:
Correction... https, not http...

2. Now you can open the web browser on your laptop and browse to https://localhost:64443 to access your TinyPilot all while not exposing your TinyPilot to the public internet.

Also.. until this bug is fixed...
3. When the TinyPilot redirect redirects to the `/login` page and fails, fix the URL redirected to to be `https` and add the `:64443` port reference.

mtlynch, 2021-02-04:
Okay, gotcha. I think this is the same root cause as this:

https://github.com/mtlynch/ansible-role-tinypilot/issues/85

I asked another customer to test the fix by running the command below, but they reported that it didn't quite fix the issue. I need to test the fix a bit more to figure out what's going wrong.

### Try the fix

```bash
sudo sed -i 's/proxy_set_header Host $host/proxy_set_header Host $http_host/g' \
  /etc/nginx/sites-enabled/tinypilot.conf && \
  sudo service nginx restart
```

### Revert back to normal

```bash
sudo sed -i 's/proxy_set_header Host $http_host/proxy_set_header Host $host/g' \
  /etc/nginx/sites-enabled/tinypilot.conf && \
  sudo service nginx restart
```

ossiemoore, 2021-02-04:
## Simple Workaround

Just append the `/login`.   If you use `https://localhost:6443/login` in my example above instead of first browsing to `https://localhost:6443/`, it works fine.

ossiemoore, 2021-02-09:
FYI.. I did actually try the fix you noted above, that did not solve the problem. That said, the workaround of just adding the "/login" to the URL resolved the issue for me.

mtlynch, 2021-02-09:
@ossiemoore thanks for letting me know!

I'm still pretty sure it's an nginx configuration issue. I just have to find a few hours to simulate an equivalent network setup so I can try to repro this and look for a fix.

closed 2021-03-08 by mtlynch