tiny-pilot/tinypilot #473: Keystroke Echo Options (none, virtual, typed, both)

ossiemoore, 2021-02-07:
### Overview and Reasoning
The browser echos all or no keystroke activity in the keystroke history area above the virtual keyboard depending on if you check/uncheck the "Show history" checkbox. This is great, but it would be even better if you could set that for the Virtual and Physical keystrokes independently. Useful, for example, to see that the virtual keyboard "clicks" are truly sending a key while preventing every keystroke you type being echoed. (No more having to remember to turn off show history when typing a password for example.)

### Feature Requested

Keyboard echo options of:
- None (don't echo any of the keystrokes)
- Virtual Only (only echo the keystrokes generated by clicking on the virtual keyboard)
- Typed Only (only echo the keystrokes typed using the keyboard)
- Both (typed or virtual)

mtlynch, 2021-02-09:
Thanks for submitting this! Can you tell me a bit more about your scenario, though? What's a situation where you want virtual keys echoed but not physical keys or vice versa? If it just showed you errors without revealing the keystrokes, would that work?

mtlynch, 2021-04-22:
Closing due to lack of response, but feel free to open with more details

closed 2021-04-22 by mtlynch