tiny-pilot/tinypilot #481: Support keyboard macros / shortcuts

mtlynch, 2021-02-10:
Several users have requested some way of saving key sequences like Ctrl + Alt + Del (for Windows) or ScrLk + 1 (for certain KVMs).

We should offer a command for executing common macros and adding user-defined key sequences.

db39, 2023-01-31:
A [user on the forum has requested this functionality](https://forum.tinypilotkvm.com/-537/adding-keyboard-shortcuts-to-menu) to enter keyboard shortcuts like ALT+TAB.

storm134, 2023-11-14:
i would like to add my own keyboard shortcuts. right now this is the biggest issue i see. we design our own in house software and hardware and some of the design require you to press and hold a key then press the desired number for what you want to do. normally not big deal the employee does this at the station. but would be great for them to be able to do it remotely. would also be great to be able to use this with a kvm. i am not opposed to buying one for every machine but that can start to take up space and power. maybe offer a tiny pilot version that cost more but works with kvm

db39, 2024-05-07:
Another user has expressed interest in this functionality via Helpscout.

zatricky, 2024-11-09:
To add, REISUB is currently impossible to execute via Tinypilot. See https://wiki.archlinux.org/title/Keyboard_shortcuts#Rebooting

cghague, 2024-11-11:
Thanks for your message, @zatricky! We have a separate issue open for [supporting the Magic SysRq key combination](https://github.com/tiny-pilot/tinypilot/issues/1167) that you might find interesting. The [current workaround](https://github.com/tiny-pilot/tinypilot/issues/1167#issuecomment-1847677077) is to run a script over SSH, but we're working on making that process easier, so please keep watching [our blog](https://tinypilotkvm.com/blog/) for updates!

japhiel, 2025-04-29:
Another user reached out through HelpScout expressing interest in this feature.

japhiel, 2025-09-05:
Another user reached out through our support channel expressing interest in this feature.