tiny-pilot/tinypilot #493: Virtual Media Support

maximveksler, 2021-02-12:
Hi,

Wondering out loud :) 

Fancy KVM's have support for what is called USB Virtual Media, in essence, the ability to load an ISO on the KVM and show it to the connected PC as if it's a Bootable USB stick that you can then boot from and install the OS.

The same feature is available in high-end servers, like https://www.dell.com/support/kbdoc/en-il/000124001/using-the-virtual-media-function-on-idrac-6-7-8-and-9

Would it be possible, and if so - how? to hack TinyPilot to be able to do this?

maximveksler, 2021-02-12:
Update https://github.com/mtlynch/tinypilot/issues/342 seems this has been requested before. In that case, the question becomes - When will it become available? :)

mtlynch, 2021-02-12:
Thanks for checking out the project! I expect to have virtual media support available in the next 4-8 weeks in TinyPilot Pro.

closed 2021-02-12 by mtlynch