tiny-pilot/tinypilot #497: Allow more control over the processing of input events

jotaen, 2021-02-12:
Consider the following cases:

- I have TinyPilot open in my browser. I now want to switch to a different app on my host machine by hitting “Meta”+“Tab”. In this case the “Meta” key stroke still gets send to the remote machine. (See first half of the attached gif.)
- When moving the mouse cursor across the host machine screen while the TinyPilot window is in the foreground, all mouse movements are send to the remote machine. If – for whatever reason – I don’t intend to change the mouse cursor position on the remote machine, I would have to navigate the mouse cursor around the browser window.
- When I click the “Power” button in the UI (top right corner) an overlay opens. My intuitive assumption was that I can make the overlay go away (a.k.a. “cancel”) by hitting the “ESC” key. This isn’t the case, but instead the “ESC” key stroke got send to the remote machine. This was unexpected, because since the whole UI is covered by the overlay I would not expect any key strokes to be sent to the remote at all. (See second half of the attached gif.)

While the last item can even be considered to be a bug, the first two are more a usability thing. It’s admittedly unlikely that this will ever cause a problem, because usually pressing “Meta” or moving the cursor alone doesn’t do anything. On the other hand this is also not guaranteed and there might be legitimitate use cases where a user wants to “mute” input devices without having to close the TinyPilot window.

A possible solution might be to give users an option to decide that, e.g. via a toggle button à la “capture input devices”. That would be obivously turned on by default, but in case a user wants to have full control, they can use that option.

By the way, VirtualBox has a similar feature, where you can [turn on and off the “input device capturing”](https://www.virtualbox.org/manual/ch01.html#keyb_mouse_normal) for a running VM. They have a so-called “Host key” to control this behaviour.

[This issue](https://github.com/mtlynch/tinypilot/issues/306) is slightly related, as it is (broadly speaking) also about the topic of what key strokes are processed under which circumstances.

[image]

jotaen, 2021-02-12:
Well actually I just noticed that the “Meta” key is not processed anyway by TinyPilot?

`WARNING in socket_api: Unrecognized key: Meta (keycode=OSLeft)`

mtlynch, 2021-02-12:
>Well actually I just noticed that the “Meta” key is not processed anyway by TinyPilot?

>`WARNING in socket_api: Unrecognized key: Meta (keycode=OSLeft`

Oh, that shouldn't be. That looks like a separate bug related to non-US keyboards.

Can you file a bug for that? It'd be helpful to include a screenshot of what you see when you hit the same key on keycode.info.

jotaen, 2021-02-12:
Bug filed: https://github.com/mtlynch/tinypilot/issues/498

(I’m using the English/International keyboard layout, btw.)

jotaen4tinypilot, 2021-08-16:
Just for alignment, my idea of this feature back in the day was roughly like so:

[image]

[image]

- Wording tbd; “Capture Inputs”, “Forward Inputs”, “Capture Mouse/Keyboard”,… not sure what is most clear.
- The user can “mute” and “activate” input forwarding via a menu item (which is toggled on by default obviously).
- If capturing (or whatever we call this) is off, we display that prominently in the status bar. (Could even be more prominent.)
- This is a client-side feature, not a system setting. It’s basically like a lightweight and local readonly mode.

Is this what you had in mind too? Or were you thinking into a different direction?

jotaen, 2021-08-16:
Just for alignment, my idea of this feature back in the day was roughly like so:

[image]

[image]

- Wording tbd; “Capture Inputs”, “Forward Inputs”, “Capture Mouse/Keyboard”,… not sure what is most clear.
- The user can “mute” and “activate” input forwarding via a menu item (which is toggled on by default obviously).
- If capturing (or whatever we call this) is off, we display that prominently in the status bar. (Could even be more prominent.)
- This is a client-side feature, not a system setting. It’s basically like a lightweight and local readonly mode.

Is this what you had in mind too? Or were you thinking into a different direction?

jotaen4tinypilot, 2021-08-16:
(Sorry for double-posting, sometimes I mix up my two accounts… 🤯 😃)

mtlynch, 2021-08-16:
That's what I had in mind, except I'm now questioning the benefit given the cost.

The first issue is that it doesn't really belong in any existing menu. "View" is for items that just change what's in the UI, so turning off input capturing seems too outside of that category. It's not really a System menu thing because it doesn't affect the System's permanent state, and it's not really an Action.

VirtualBox needs this feature because otherwise your input is stuck in the VM. You can't move your mouse cursor out or type in other windows until you turn off input capturing. We don't have that issue.

I haven't run into a ton of situations where it would be important for me to disable input capturing. The Meta+Tab issue has happened, but it's not so irritating that I'd first navigate to the TinyPilot menu to turn off input capturing and then turn it back on.

I'm also worried about it becoming a support burden. There are situations where input forwarding fails, so having to do an extra step of, "Can you check in the UI to verify that input capturing isn't disabled," would be a pain.

Can you think of scenarios where this feature provides high value?

jotaen, 2021-08-17:
> "View" is for items that just change what's in the UI, so turning off input capturing seems too outside of that category.

Yeah, agreed it doesn’t really fit in there. I guess putting it into “View” would be more like the *least bad* option for the lack of an optimal alternative. I also agree it should even less go into “Actions” or “System”.

> Can you think of scenarios where this feature provides high value?

The initial context where I brought this up was when the modifier keys could get stuck on the target machine. This issue is fixed by now, so that motivation has become obsolete. Apart from that, I can imagine this to be useful as a sort of readonly mode, e.g. when the user wants to observe the remote computer but wants to avoid generating inputs by accident.

I find it pretty hard to speculate, though, whether people would appreciate having more control like this or not – I can imagine both ways. It’s certainly not a killer feature, I’d consider it more like a nice-to-have that makes the interface a bit more powerful. I’m also not sure whether this would become obsolete by a server-side readonly mode / interface, or whether it still would be useful in its own right.

mtlynch, 2021-08-17:
Yeah, let's close this as wontfix for now and reopen if we run into situations where it would be useful. I think the guest mode feature should cover instances where we'd want this.

closed 2021-08-17 by mtlynch