tiny-pilot/tinypilot #501: Allow TinyPilot to call privileged scripts in dev mode

mtlynch, 2021-02-12:
*\[splitting off from a comment in [#433](https://github.com/mtlynch/tinypilot/issues/433#issuecomment-778423702)\]*

Currently, in a development environment, the scripts under `/opt/tinypilot-privileged` don't exist, so API endpoints that depend on those scripts fail. This causes drift between the production and development environment, making it easier for bugs that affect the prod environment to slip through development unnoticed.

### Option 1: Add mocks

When the user starts the server in dev mode (maybe we'll add an option for specifying this), the paths to privileged scripts get replaced with paths to dummy scripts in the repo that simulate behavior of the real script without causing any external effect.

### Option 2: Add an ansible example command for dev-mode installation

It should be possible to specify the right commands to the Ansible role so that it installs TinyPilot in dev mode, but with the real production privileged scripts. This might not be such a good solution for a general dev machine, but it would work well if you do development in a dedicated TinyPilot VM.

jotaen, 2021-03-10:
I’d really like to emphasize how much I would appreciate this. Maybe it’s just me, but the UI failing in weird ways locally feels very strange to me.

Regarding the options: the scripts path could be configurable (e.g. through an env-variable), so by default we point to a folder with mock scripts, but you can overwrite it if you are in an environment where you want to use the real ones.

mtlynch, 2021-03-10:
I'm reluctant to add dev-only logic to the production code. I'm not totally against it, but I'd like to avoid it if possible.

Would it be viable to have a mock-scripts directory in the tinypilot repo and then part of the dev installation is symlinking from `/opt/tinypilot-privileged` -> `dev-scripts/mock-scripts`?

jotaen, 2021-03-10:
Ah sorry, I meant the other way around regarding the env variable: the default would be `/opt/tinypilot-priviledged` of course, but you can make it read from `dev-scripts/mock-scripts` by setting an env variable explicitly.

A symlink would also be perfectly viable.

mtlynch, 2021-03-10:
>Ah sorry, I meant the other way around regarding the env variable: the default would be /opt/tinypilot-priviledged of course, but you can make it read from dev-scripts/mock-scripts by setting an env variable explicitly.

Oh, that's what I understood. I just mean we'd have to make code changes that dynamically resolve the path depending on the env variable, so I'd rather avoid that as there's (low) potential for dev-only code to inadvertently mess up prod scenarios.

The UI changes are higher priority to me, but I'll take this.

jotaen, 2021-03-16:
@mtlynch is this done or do you want to keep it open for the remaining scripts?

mtlynch, 2021-03-16:
I was leaving it open since it's not *fully* fixed since we don't mock out the upgrade script, but I guess we should close it for now and reopen if the upgrade issue becomes an obstacle.

closed 2021-03-16 by mtlynch