tiny-pilot/tinypilot #522: In fullscreen mode, escape key exits fullscreen mode

mtlynch, 2021-02-18:
## Description

A user [reported](https://forum.tinypilotkvm.com/-15/possible-to-remap-the-esc-key#post-1) that hitting the escape key while in fullscreen mode causes fullscreen mode to turn off rather than just forwarding the key to the target machine as an "Escape" key.

## What's the behavior that you expect?

TinyPilot forwards the escape key to the target machine and remains in fullscreen mode.

## What's happening instead?

Hitting the "Escape" key causes TinyPilot to exit fullscreen mode.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on Commands > Fullscreen
3. Hit "Escape"

mtlynch, 2021-02-26:
One consideration here is how users exit fullscreen if we start absorbing the Escape key. Escape works cross-browser, so if we take that away, it might become hard to explain to users how to exit fullscreen.

Maybe we should treat two consecutive Escapes as "real" exit fullscreen.

djclueless, 2021-12-29:
Here is a work around but... This solution will not work with every browser and it does not seem to work on non-secure sites.

Below is a detailed explanation of a work around which would require holding down `ESC` to exit full screen. 

[image]

https://developer.mozilla.org/en-US/docs/Web/API/Keyboard/lock

https://wicg.github.io/keyboard-lock/#escape-key

```
# Lock keyboard and go full screen
navigator.keyboard.lock();
document.documentElement.requestFullscreen();
```

```
# Exit full screen and unlock keyboard.
document.exitFullscreen();
navigator.keyboard.unlock();
```

[image]

* attempted to test on Chrome 96.0.4664.110 on a non secure TinyPilot.  navigator.keyboard was `undefined`.

mtlynch, 2021-12-29:
Thanks for investigating this! The `keyboard.lock` feels a bit too brittle at this point. It's strange that it doesn't work under HTTPS, since they don't seem to document that.