tiny-pilot/tinypilot #533: Handle HTTPS in hostname change dialog

mtlynch, 2021-02-25:
I realized that there are some wrinkles in changing the hostname when HTTPS is involved:

1. When we change the hostname with HTTPS enabled, we need to generate a new TLS key and cert with the new hostname assigned (see [this PR](https://github.com/tiny-pilot/ansible-role-tinypilot-pro/pull/32/files)). This process adds ~20s to the change hostname script.
2. If the user hasn't installed the device's CA certificate, `/status` requests to the new hostname will fail due to TLS errors because the new domain is untrusted.

For (1), I think there are two possibilities:

* Change `/api/hostname` to use the websockets channel. I *believe* WebSockets is more suited to long-running requests, though I'm not positive.
* Change `/api/hostname` to an asynchronous API, similar to `/api/update`
  * It wouldn't have to be as complicated as `/api/update` because that API has to account for the whole backend restarting between requests, whereas the hostname change won't restart the backend

For (2), can we distinguish between a failed network connection and a failed TLS connection? If not, we should detect if the new address is HTTPS and suggest that as a reason for the failure to auto-redirect.

mtlynch, 2021-02-25:
Actually, false alarm. I realized there's a simpler solution.

Instead of regenerating the TLS key immediately, we can create a systemd service that does it on next boot. That should keep the `/api/hostname` request fast (maybe 1s of delay). (2) is still an issue, but it's much less of a priority in isolation. I'll track that in a separate bug.

closed 2021-02-25 by mtlynch

jotaen, 2021-02-25:
👍 this solution coincides with how the hostname change comes into effect anyway (i.e. on next reboot, and not immediately).