tiny-pilot/tinypilot #557: scripts/uppgrade fails on 'get TinyPilot repo'

matthewdavis, 2021-03-08:
## Description

Updater script fails with

```
TASK [tinypilot.tinypilot-pro : get TinyPilot repo] ****************************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": ["/usr/bin/git", "fetch", "--tags", "origin"], "msg": "Failed to download remote objects and refs:  From https://dev.azure.com/tinypilot/tinypilot-pro/_git/tinypilot-pro\n ! [rejected]        2.0.0      -> 2.0.0  (would clobber existing tag)\n"}
```

mtlynch, 2021-03-08:
Sorry about that! I just pushed a fix. Can you retry?

matthewdavis, 2021-03-08:
Fixed. Thanks for the quick turnaround!

```
TASK [tinypilot.tinypilot-pro : get TinyPilot repo] ****************************************************************************************************************************************************************************************
changed: [localhost]

TASK [tinypilot.tinypilot-pro : find absolute path to python3] *****************************************************************************************************************************************************************************
ok: [localhost]

TASK [tinypilot.tinypilot-pro : save absolute path to python3] *****************************************************************************************************************************************************************************
ok: [localhost]

TASK [tinypilot.tinypilot-pro : create TinyPilot virtualenv] *******************************************************************************************************************************************************************************
changed: [localhost]

TASK [tinypilot.tinypilot-pro : fix TinyPilot folder permissions] **************************************************************************************************************************************************************************
changed: [localhost]

TASK [tinypilot.tinypilot-pro : install TinyPilot as a service] ****************************************************************************************************************************************************************************
ok: [localhost]

TASK [tinypilot.tinypilot-pro : enable systemd TinyPilot service file] *********************************************************************************************************************************************************************
ok: [localhost]

TASK [tinypilot.tinypilot-pro : install tinypilot-updater as a service] ********************************************************************************************************************************************************************
ok: [localhost]

RUNNING HANDLER [tinypilot.tinypilot-pro : restart TinyPilot service] **********************************************************************************************************************************************************************
changed: [localhost]

PLAY RECAP *********************************************************************************************************************************************************************************************************************************
localhost                  : ok=71   changed=6    unreachable=0    failed=0    skipped=28   rescued=0    ignored=0   

pilot@tinypilot:~ $ 
```

closed 2021-03-08 by matthewdavis

mtlynch, 2021-03-08:
Thanks for reporting it! I'm getting ready to publish a blog post and make big announcements about this new version, so I'm glad you told me about this bug before huge numbers of users ran into this issue.

Also, this should be the last version that you have to update from the command-line, as you can now do it from System > Update in the web UI.