tiny-pilot/tinypilot #571: Sizing options for remote screen

jotaen, 2021-03-11:
Currently there is no way to adjust the remote screen size independtly from the browser window. It would be cool if you could have some predefined options, which could live somewhere under the “View” menu. E.g. something like:

- `Original` (original size, regardless of viewport)
- `Fit` (always make remote screen fit into viewport)
- `Fixed` (Some fixed ratio)

jotaen, 2021-06-08:
One question here: is the remote screen always shown in 640x480 or are these Macro-Silicon-HDMI chips also capable of delivering higher resolutions? I always just assumed there were higher resolutions possible as well, but I realized that might not be true?

mtlynch, 2021-06-08:
The MacroSilicon dongles capture up to 1080p. 640x480 is super small. Does your target system allow you to adjust the resolution up any more or does it only show 640x480p as an option?

jotaen, 2021-06-09:
Ah thanks, I need to check that again next week and look what settings there are (I don’t have the dongle with me currently). So far I didn’t bother much because I only use plug in the video for occasional testing.

(I have [this one](https://www.amazon.com/dp/B08C9FCF2X), btw., which from the specs should support higher resolutions.)