tiny-pilot/tinypilot #574: Video lag when not enough bandwidth

ankineri, 2021-03-11:
## Description

When there is not enough bandwidth to pass the stream through, the video lag accumulates. A page refresh helps reset the lag. 

## What's the behavior that you expect?

Dropping frames when bandwidth does not allow for full stream. Or degrade quality.

## What's happening instead?

The lag increases to values like tens of seconds.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Switch network profile to "Slow 3G"
4. Do some actions in the KVM
5. Observe the lag increase

## Logs
Note that to reduce the bandwidth I've lowered ustreamer settings (resolution, fps, quality).
http://sprunge.us/UF2fyc

mtlynch, 2021-03-11:
Thanks for reporting. This is a known issue. It's due to the fact that TinyPilot uses MJPEG for video streaming, which browsers support poorly.

I'm looking at options for switching to H264 which is much more bandwidth-efficient and has better browser support, but it's going to take a few months, since it's a significant change.

ankineri, 2021-03-15:
The author of uStreamer has mentioned that they're implementing h264 - is that the process you're referring to, or is it a separate effort?

mtlynch, 2021-03-17:
It's still up in the air whether uStreamer's h264 implementation will work with TinyPilot. It will be great if we can continue sharing it, but if not, we'll have to move to a different implementation.

marwatk, 2021-06-02:
Do you know where the buffer builds up? Without looking at all I'm guessing it's ustream, but it could just as easily be nginx, I suppose. I'd be willing to work an an intermediary layer that would detect buffer buildup and proactively drop source frames, if you'd be interested.

mtlynch, 2021-06-03:
I'm not sure, but I think the issue is on the client side in the browser. I don't think it's worth building an intermediate layer, because h264 support is expected soon, though I thought that six months ago too.

The workaround in the meantime is to reduce video settings so that TinyPilot sends frames low enough in bandwidth to avoid clogging the network.

michaelguild13, 2022-03-21:
@mtlynch any updates to switch to `h264`

mtlynch, 2022-03-21:
@michaelguild13 - We're actively working on it. You can follow updates on #875.

It's possible to test a [proof of concept install](https://github.com/tiny-pilot/tinypilot/tree/experimental/h264#install). I recommend doing it on a spare microSD, as it will overwrite any existing TinyPilot installation.

mtlynch, 2023-02-16:
This should be fixed now that H.264 is available.

closed 2023-02-16 by mtlynch