tiny-pilot/tinypilot #582: Custom HTML elements don't import their own dependencies.

jdeanwallace, 2021-03-16:
## Description

There is a sneaky import bug in the frontend code. The only place where [`controllers.js`](https://github.com/mtlynch/tinypilot/blob/master/app/static/js/controllers.js) is imported is within the [`shutdown-dialog`](https://github.com/mtlynch/tinypilot/blob/master/app/templates/custom-elements/shutdown-dialog.html#L89) custom element, and every other custom element relies on that import.

## What's the behavior that you expect?

Every custom element should import its own dependencies.

## What's happening instead?

If the `shutdown-dialog` custom element does not get used in the HTML document then [`controllers.js`](https://github.com/mtlynch/tinypilot/blob/master/app/static/js/controllers.js) will not be imported.

## What are the steps to reproduce this behavior?

1. Comment out the [`controllers.js` import from within the `shutdown-dialog` element](https://github.com/mtlynch/tinypilot/blob/master/app/templates/custom-elements/shutdown-dialog.html#L89)
2. Attempt to use any other custom element that relies on a controller function. For example, click "System" > "View Logs"
3. See the `Uncaught TypeError: controllers.getDebugLogs is not a function` error in the console.

jotaen, 2021-03-17:
I also happened to run into this already, but with CSS files in my case. When working on the redesign I already made sure to add `@import` directives to the HTML components when they rely on global CSS classes or variables, because I encountered the same problem there.

I think this will *probably* be addressed when we pick up [the frontend bundling](https://github.com/mtlynch/tinypilot/issues/538), because we go all-in on ES modules then and that forces us to reference JS dependencies explicitly rather than relying on the global initialisation order. For CSS it’s probably a bit trickier to come up with a robust solution though, since it’s all global either way.

mtlynch, 2021-07-19:
Fixed in #747