tiny-pilot/tinypilot #597: Detect last update in a more robust way

mtlynch, 2021-03-18:
The end-user needs to know when a TinyPilot update is in progress and when it's complete. This is a bit tricky because we want to keep the server stateless, so the states of "completed successfully" and "never ran at all" can look similar.

TinyPilot currently solves this with ["update result files."](https://github.com/mtlynch/tinypilot/blob/3a61e911901f976e6302a9f1badb87c32a1ce9ed/app/update_result.py) When an update completes, the update script writes a result to `/home/tinypilot/logs/YYYY-MM-DDTHHMMSSZ-update.result.json` containing a timestamp and success indicator. TinyPilot uses the timestamp to decide whether an update completed recently. If it completed within the last 8 minutes, TinyPilot assumes it was the most recent update and declares the update complete.

This is a flaky solution, as the update time is not 100% reliable.

8 minutes is a also a hack to fix a bug. It was supposed to be 3 minutes since completion, but due to a bug, the update service was writing the timestamp as the update *start* time instead of the update *finish* time, so I added 5 minutes of slop so that it didn't break updates.

But now the 8 minutes means that if you ever have to do two sequential updates, I think the logic will screw up because when you start the second update, if the update service hasn't written an initial result file yet, TinyPilot will see the last update file and think it's for *this* update because it finished "recently" (within the last 8 minutes).

## Possible solutions

### Store the git version

* Store the git version in the update result in a field like `updated_from`
  * The server knows its current version, so if it sees the `updated_from` version matches its own version, then it's the correct result for this update. If the version doesn't match, it's a stale result from a previous update.
  * Result files generated prior to this change won't have the `updated_from` field, but that could be fine because comparing `None` to `1.2.3` gets us the correct result (must be update from a different version).

#### Example

| Current server version | `updated_from` in most recent result file | Meaning |
|--------------------------|---------------------------------------------|-----------|
| `1.2.3`                            | `None`                                                      | The update is complete because the previous server version wasn't tracking server version. |
| `1.2.3`                            | `1.2.2`                                                       | The update is complete because it brought us to a higher version. |
| `1.2.3`                            | `1.2.3`                                                       | Can this happen? Need to think more....

### Parse the syslog

* Parse the update syslog and figure out the recent result that way
  * Feels kind of hacky

jotaen, 2021-04-13:
Just out of curiosity: what was the reason to support running multiple updates? Because only allowing for a single update to occur at a time would have been my naive assumption.

One edge case that comes to mind would be that for some reason the update never completes (for whichever reason), and then any future updates are locked. There could be e.g. a timeout or some manual override.

mtlynch, 2021-04-13:
>Just out of curiosity: what was the reason to support running multiple updates? Because only allowing for a single update to occur at a time would have been my naive assumption.

Oh, good question. There can't be multiple updates in parallel. The updates should only happen in sequence. But updates could happen close together, like if a user updates to version N and then during their update, N + 1 is released, so if they check for updates immediately after, they'd be able to update to N + 1.

The client could theoretically send two `/update` PUT requests in sequence, so TinyPilot would attempt to update twice, but it wouldn't actually cause two updates in parallel; it would just cause `sudo service tinypilot-updater start` to execute twice, but only one service can run at a time.

>One edge case that comes to mind would be that for some reason the update never completes (for whichever reason), and then any future updates are locked. There could be e.g. a timeout or some manual override.

Oh, that's true. If the update process hangs indefinitely, it would lock out other updates. Currently, the user would have to reboot their device to kill the old update process, but a timeout would be better. I've filed a bug in the ansible role to track:

https://github.com/mtlynch/ansible-role-tinypilot/issues/130

closed 2021-04-23 by mtlynch