tiny-pilot/tinypilot #598: libssl-dev : Depends: libssl1.1 - unable to install

dharrigan, 2021-03-19:
```
❯ uname -a
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

❯ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

❯ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.17-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP Mon Feb 22 11:33:35 GMT 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
```

## Description

Whilst trying to install tinypilot, an error is thrown which stops installation.

## What's the behavior that you expect?

Installation completes.

## What's happening instead?

```
The following packages have unmet dependencies:
 libssl-dev : Depends: libssl1.1 (= 1.1.1d-0+deb10u5) but 1.1.1d-0+deb10u5+rpt2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

## What are the steps to reproduce this behavior?

1. Install latest Raspbian on Raspberry PI 4.
2. Ensure all is up-to-date by running `apt-get update` and `apt-get upgrade -y`
3. Run:

```
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

## Logs

```
...
...
Reading package lists... Done                          
+ sudo apt-get install -y git libffi-dev libssl-dev python3-dev python3-venv
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u3).
python3-dev is already the newest version (3.7.3-1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libssl-dev : Depends: libssl1.1 (= 1.1.1d-0+deb10u5) but 1.1.1d-0+deb10u5+rpt2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

mtlynch, 2021-03-19:
Thanks for reporting this!

Is this reproducible on a clean RasPi OS install? It looks like there may be something wrong with your package manager on that particular system.

>E: Unable to correct problems, you have held broken packages.

It's also possible that some of the apt packages changed and the dependencies need to be updated, but if you could try it on a clean system, that would be helpful.

dharrigan, 2021-03-19:
Hi,

It was a fresh install, brand new from latest image - installed about 30 mins before I attempted to install tinypilot. Admittingly, it's not a completely untainted installation, as I normally bring my system up-to-date with my standard user configuration (using ansible) before I do anything else.

I did install tinypilot on the same machine, last week without issue, from again a fresh base installation (I wiped the card this morning to try something out, so rebuilding things again).

I can attempt again with another fresh install, before I do my user ansible setup.

dharrigan, 2021-03-19:
Hi,

Frest install, no issues detected. I suppose something that I install prior gets in the way. Happy to close this issue as not a fault of tinypilot. :-)

closed 2021-03-19 by dharrigan

mtlynch, 2021-03-19:
Thanks for letting me know!