tiny-pilot/tinypilot #600: Target Hosts Failing to Enumerate Properly

ryanohoro, 2021-03-21:
## Description

Target system not properly recognizing Tinypilot HID devices. Fresh RPi 4, Fresh Raspberry Pi OS image, fresh install from `ansible-role-tinypilot`, **multiple USB-C to USB-A cables tested**, multiple target systems. Logs are from another RPi 4.

I followed the troubleshooting steps in the Wiki.

(Thanks! Tinypilot is a great project, I hope to use it more!)

## What's the behavior that you expect?

Target host enumerates the TinyPilot HID device properly

## What's happening instead?

Target ​host recognizes that the Tinypilot devices are connected, but throws errors.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on '....'
4. See error

## Screenshots

*If applicable, add screenshots to help explain your problem.*

## Logs

Tinypilot

```
$ dmesg
[  168.622397] dwc2 fe980000.usb: new device is high-speed
[  168.721803] dwc2 fe980000.usb: new device is high-speed
[  168.811686] dwc2 fe980000.usb: new address 2
[  168.846094] configfs-gadget gadget: high-speed config #1: c
```

Tinypilot - intermittent error
```
$ dmesg
[ 1341.750465] configfs-gadget gadget: End Point Request ERROR: -108
[ 1341.750490] configfs-gadget gadget: End Point Request ERROR: -108
[ 1341.760482] dwc2 fe980000.usb: new device is high-speed
[ 1341.859870] dwc2 fe980000.usb: new device is high-speed
[ 1341.949751] dwc2 fe980000.usb: new address 2
[ 1341.983776] configfs-gadget gadget: high-speed config #1: c
```

Target
```
$ dmesg
[  310.857489] hid-generic 0003:1D6B:0104.0004: collection stack underflow
[  310.857582] hid-generic 0003:1D6B:0104.0004: item 0 0 0 12 parsing failed
[  310.857706] hid-generic: probe of 0003:1D6B:0104.0004 failed with error -22
[  310.859872] hid-generic 0003:1D6B:0104.0005: collection stack underflow
[  310.859947] hid-generic 0003:1D6B:0104.0005: item 0 0 0 12 parsing failed
[  310.870118] hid-generic: probe of 0003:1D6B:0104.0005 failed with error -22
```

 `/opt/tinypilot/dev-scripts/dump-logs`
 
[tinypilot-error-bug-report.txt](https://github.com/mtlynch/tinypilot/files/6178295/tinypilot-error-bug-report.txt)

CoreParadox, 2021-03-21:
Maybe related to my issue, I've only tested with windows hosts though. 
https://github.com/mtlynch/tinypilot/issues/599

mtlynch, 2021-03-22:
Sorry about this! I pushed [a bad change](https://github.com/mtlynch/ansible-role-tinypilot/pull/115/files) on Friday that broke input forwarding. It's fixed now. If you update to the latest version, it should restore input forwarding.

closed 2021-03-22 by mtlynch

ryanohoro, 2021-03-22:
Oh my. Thanks so much! I thought I was going batty not being able to get it work.

ryanohoro, 2021-03-22:
UI update option says it's up to date. It looks like the latestRelease API only checks the latest commit for `tinypilot` and not `ansible-role-tinypilot`, which won't resolve this bug obviously.

I manually re-ran the bootstrap command, and `/sys/kernel/config/usb_gadget/g1/functions/hid.mouse/report_desc` ends up still full of garbage.

I manually re-ran `/opt/tinypilot-privileged/init-usb-gadget` and got an error which had not appeared in the ansible output. The script may not emit a non-zero exit on error?

Output

```
$ sudo /opt/tinypilot-privileged/init-usb-gadget
+ set -u
+ modprobe libcomposite
+ cd /sys/kernel/config/usb_gadget/
+ mkdir -p g1
+ cd g1
+ echo 0x1d6b
+ echo 0x0104
+ echo 0x0100
+ echo 0x0200
+ STRINGS_DIR=strings/0x409
+ mkdir -p strings/0x409
+ echo 6b65796d696d6570690
+ echo tinypilot
+ echo 'Multifunction USB Device'
+ KEYBOARD_FUNCTIONS_DIR=functions/hid.keyboard
+ mkdir -p functions/hid.keyboard
+ echo 1
/opt/tinypilot-privileged/init-usb-gadget: line 36: echo: write error: Device or resource busy
```

Line

```
echo 1 > "${KEYBOARD_FUNCTIONS_DIR}/protocol" # Keyboard
```

ryanohoro, 2021-03-22:
Running `/opt/tinypilot-privileged/remove-usb-gadget` before running the bootstrap command works around the error, and I can confirm the revert-fix is working.

mtlynch, 2021-03-22:
>UI update option says it's up to date. It looks like the latestRelease API only checks the latest commit for tinypilot and not ansible-role-tinypilot, which won't resolve this bug obviously.

Hmm, that's strange that it claimed it was up to date. The updater only checks the tinypilot repo, but I [pushed a change](https://github.com/mtlynch/tinypilot/commit/1a77069bcda9413f79c8b12ec50e97bb854ee35b) before you posted, so the updater should have updated based on that. Then the update script automatically downloads the latest version of ansible-role-tinypilot.

It sounds like you were able to get it working with some tinkering. Sorry for the hassle! Let me know if you run into any other issues.

ryanohoro, 2021-03-22:
I see. Looks like I ran the bootstrap command first (before checking the UI update method).

The bootstrap command failed to fix the underlying problem because of the init-usb-gadget write error. The commit hash was likely updated and the silent failure ignored during the update. At that point the UI update method was showing up to date, but the affected files from the bug fix weren't overwritten properly.