tiny-pilot/tinypilot #603: Documentation Clarifications

cdgriffith, 2021-03-22:
1. https://github.com/mtlynch/tinypilot/wiki/Troubleshooting-and-Diagnostics#browser-console-logs  shouldn't suggest to use `F12` to open the console, as that is captured for input

2. https://github.com/mtlynch/tinypilot/wiki/Troubleshooting-and-Diagnostics#browser-console-logs  does not specify where to update the environment variables, should add they are in ` /lib/systemd/system/tinypilot.service` 

3. Not clearly mentioned of where tinypilot is installed by default: `/opt/tinypilot`    - would suggest adding to https://github.com/mtlynch/tinypilot#simple-installation

mtlynch, 2021-03-23:
Thanks for reporting these!

>https://github.com/mtlynch/tinypilot/wiki/Troubleshooting-and-Diagnostics#browser-console-logs shouldn't suggest to use F12 to open the console, as that is captured for input

Good point, fixed!

>Not clearly mentioned of where tinypilot is installed by default: /opt/tinypilot - would suggest adding to https://github.com/mtlynch/tinypilot#simple-installation

True. Fixed in #608.

>https://github.com/mtlynch/tinypilot/wiki/Troubleshooting-and-Diagnostics#browser-console-logs does not specify where to update the environment variables, should add they are in /lib/systemd/system/tinypilot.service

Is that the correct URL? I don't see the mention of environment variables you're talking about in that section.

cdgriffith, 2021-03-24:
Whoops second one was a bad double paste. Under the Options on the Readme for environmental vars.

PR #621 (2021-03-24): Move documentation of environment variables to CONTRIBUTING.md
They aren't really something an end-user should change. They are more useful to developers working on the code.

For end-users, the preferred way to change options is through the settings.yml file and advanced installation (linked to in the README).

Fixes #603
closed 2021-03-24 by mtlynch