tiny-pilot/tinypilot #610: Setting tinypilot to portrait mode

paullee2018, 2021-03-24:
Hi,

Currently my monitor is setup in portrait mode, how would set the tiny pilot web interface to work in this mode?

Paul

mtlynch, 2021-03-25:
Unfortunately, it's not possible currently.

I'm not sure if the capture devices even support portrait video modes. I can keep it as an open issue for future investigation.

thomstratton, 2022-04-16:
Love project!  Same Issue for me and my hardware does not support portrait resolution... but figured out a work-around using some CSS.  This is only a temp fix and your mileage may vary!  
Also this is nowhere near complete it will probably get wiped out when you update tinypilot... just a proof of concept that could provide a solution once if added into the project properly.

if you are comfortable editing code:

- back up /opt/tinypilot/app/templates/index.html
- edit /opt/tinypilot/app/templates/index.html

add the inline style line below to the #"remote-screen" element
 
it should looks something like this - you may need to adjust the height and margins for your screen.  The key part is:
`transform:rotate(0.25turn);`

~~~html
<remote-screen
    id="remote-screen"
    style="height: 65vh;margin-top: 10vh;margin-bottom: 20vh;transform: rotate(0.25turn);"
    milliseconds-between-mouse-events="600"
>
~~~ 

restart your pi... the remote screen should now be flipped on the page.  If it is upside down for you, change the value 0.25 to 0.75.

and adjust margins / height / width in css as needed.

good luck!

thomstratton, 2022-04-16:
also I should mention I have only tested in chromium on the pi itself.