tiny-pilot/tinypilot #613: Unstable video

Kostecki, 2021-03-24:
## Description

I get frequent flickering between showing the video output and the "no signal" screen. It'll show video, switch to "no signal" for a moment and then back to video. Repeating forever.

Sometimes it'll even throw up in the console and kill video output altogether

## What's the behavior that you expect?

For TinyPilot to show video output from my server. Indefinitely.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. See error

## Screenshots

*If applicable, add screenshots to help explain your problem.*

## Logs
TinyPilot: https://logs.tinypilotkvm.com/gbVG7ayG
Kernel errors: https://pastebin.com/wyCJ2eqd

mtlynch, 2021-03-24:
Thanks for reporting this. What device are you using for video capture?

Kostecki, 2021-03-24:
The generic ebay/aliexpress small form factor hdmi capture device

mtlynch, 2021-03-25:
I think this is a failure in your capture device. The USB to HDMI dongles are low-cost, but they're not the most reliable. It seems like about 2-3% of them are defective. Would it be possible for you to try a different dongle? I like [these](https://smile.amazon.com/dp/B08CXWPYQ8/), but they're also available on AliExpress.

Feel free to reopen if you see this reproduce on another dongle.

closed 2021-03-25 by mtlynch

Kostecki, 2021-03-25:
I'll grab one of those and give it a try 👍