tiny-pilot/tinypilot #620: Modifier keys don't work in combination with mouse clicks

mtlynch, 2021-03-24:
## Description

TinyPilot doesn't forward input events correctly for an event like Shift + left click.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Open a text field on the remote screen
3. Hold shift and click different spots in the text

## What's the behavior that you expect?

The target computer recognizes a shift + click and highlights the text

## What's happening instead?

No text is highlighted

## Example

https://user-images.githubusercontent.com/7783288/112372270-f4dc9c80-8cb5-11eb-89f4-529efaa88997.mp4

mtlynch, 2021-03-25:
A user [reports](https://forum.tinypilotkvm.com/-32#post-7) that this was working in [1.3.0](https://github.com/mtlynch/tinypilot/releases/tag/1.3.0) but broke after that. I haven't tested a 1.3.0 build to reproduce the working behavior.

jdeanwallace, 2021-05-07:
> A user [reports](https://forum.tinypilotkvm.com/-32#post-7) that this was working in [1.3.0](https://github.com/mtlynch/tinypilot/releases/tag/1.3.0) but broke after that. I haven't tested a 1.3.0 build to reproduce the working behavior.

I tested with tag 1.3.0 and I still can't reproduce the working behavior. Here's an underwhelming [demo video](https://www.loom.com/share/774e26be48d1467bb62d99370e1f4337).

mtlynch, 2021-05-12:
I think we're forwarding the shift key event incorrectly. 

If the user presses the left shift key on its own, TinyPilot currently sends:

```
0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00
^^^  [       All other keys zero'ed        ]
 |
 |
 |
 Left shift modifier key
```

What we probably want is:

```
0x02 0xe1 0x00 0x00 0x00 0x00 0x00 0x00
^^^^ ^^^^  [  All other keys zero'ed   ]
 |     |
 |   Left shift key code
 |
 Left shift modifier key
```

Key code from HID spec, but also [here](https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2#file-usb_hid_keys-h-L258)

I tried sending the raw HID code directly from the command line to simulate holding down right shift and TinyPilot had the desired shift+click behavior when I used the mouse:

https://user-images.githubusercontent.com/7783288/117991413-046a8000-b30c-11eb-8dca-0b734942a501.mp4

I haven't tested it yet, but I recommend connecting a real, physical keyboard to a test machine and running [usbhid-dump](https://www.systutorials.com/docs/linux/man/8-usbhid-dump/), then see what HID data a real keyboard sends when the user hits a modifier key in isolation. The part I'm not sure about is whether the keyboard still sets modifier bits (the first byte of the HID message) when you type a modifier key by itself.

Then, we'll need to update the HID logic on TinyPilot to match the behavior of a real keyboard for shift and all other modifiers.

jdeanwallace, 2021-05-12:
Thanks for the advice. I tested using a physical keyboard & mouse connected to the host machine. Here's a [video](https://www.loom.com/share/7782c6d61d0e4ba3b9027a05cfff6dda) of that.

Summary:
1. A physical keyboard also doesn't set the key code (when shift is pressed). It only sets the control key. So TinyPilot is doing the right thing. The problem is that TinyPilot always releases the key after it's been pressed. The problem code [here](https://github.com/mtlynch/tinypilot/blob/master/app/hid/keyboard.py#L17-L20) should also check if a control key has been pressed.
1. A physical mouse only writes 4 bytes, but TinyPilot writes 7 bytes. When streaming the HID input it looks like the buffer is overflowing (it seems like it's expecting a maximum of 5 bytes, then wraps the extra 2 bytes). Do you have a reference to the protocol we're using?
1. (not shown in the video) When fixing #1, I'm able to press shift via the TinyPilot web UI and click via the host machine mouse. This results in an expected shift+click behaviour.

mtlynch, 2021-05-12:
>A physical keyboard also doesn't set the key code (when shift is pressed). It only sets the control key. So TinyPilot is doing the right thing. The problem is that TinyPilot always releases the key after it's been pressed. The problem code here should also check if a control key has been pressed before releasing the key.

Ah, interesting! Looks like that's the issue.

>A physical mouse only writes 4 bytes, but TinyPilot writes 7 bytes. When streaming the HID input it looks like the buffer is overflowing (it seems like it's expecting a maximum of 5 bytes, then wraps the extra 2 bytes). Do you have a reference to the protocol we're using?

The report size varies by mouse, as different mice offer different features. TinyPilot's mouse is a little wacky because it's an absolute mouse (it reports absolute position on the screen, like a drawing tablet). Most physical mice are relative mice, meaning that when you move them left 1cm, they move the cursor left by however many pixels. But TinyPilot needs to specify an absolute position (when the mouse is over point x,y on the remote image, TinyPilot needs to move the remote cursor to position x', y' on the remote screen, scaling for differences in screen size).

We specify the HID descriptor here:

https://github.com/mtlynch/ansible-role-tinypilot/blob/ed29bacf132625c26b24898c7a08376ba3d22c61/files/init-usb-gadget#L81

We declare a `report_length` of `5`, so I think you're right about writing too many bytes.

The best resources I've found about the HID descriptor are these posts:

* [Tutorial about USB HID Report Descriptors](https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/)
* [Using RPi Zero as a Keyboard (3-part series)](https://www.rmedgar.com/blog/using-rpi-zero-as-keyboard-setup-and-device-definition/)

It'd be good to confirm from reading the docs that 5 bytes is correct and then make a separate fix for that.

PR #692 (2021-05-18): Fix modifier keys that don't work in combination with mouse clicks.
`hid.keyboard.send_keystroke` was always releasing the key after it has been pressed. This is because a keycode is not actually set when a modifier is pressed on it's own. This stopped us from doing things like holding in the shift key and clicking with the mouse to highlight text. [Demo video](https://www.loom.com/share/8a4629a3a70f459d9869cd493e04289a).

Fixes #620

---
This change is [[image]](https://reviewable.io/reviews/mtlynch/tinypilot/692)
closed 2021-05-18 by jdeanwallace