tiny-pilot/tinypilot #658: Keyoard doesn't work with custom domain

lamarios, 2021-04-20:
## Description

When using a custom domain https://kvm.mydomain.com the keyboard connection doesn't work anymore, the rest of the UI / ustreamer window works fine. It used to work.

## What's the behavior that you expect?

Keyboard input to be working

## What's happening instead?

Browser error:
```
 POST https://kvm.mydomain.com/socket.io/?EIO=4&transport=polling&t=NZizefx&sid=y6UV9qZRc9RF-
WebSocket connection to 'wss://kvm.mydomain.com/socket.io/?EIO=4&transport=websocket&sid=y6UV9qZRc9RF-6sfAAAn' failed: 
```

## What are the steps to reproduce this behavior?

1. Set a custom domain for tinypilot (https://kvm.mydomain.com)
2. Set up a reverse proxy to redirect domain to tinypilot. (HAProxy in my case)
2. Open tinypilot with this domain
4. Keyboard not working

mtlynch, 2021-04-20:
Thanks for reporting! Can you share your [logs](https://github.com/mtlynch/tinypilot/wiki/Troubleshooting-and-Diagnostics#logs) and HAProxy config?

lamarios, 2021-04-21:
Forgot to mention but if i use the ip of the machine running tinypilot it works as expected and it is installed on an ubuntu machine with the install script.

Logs: 
```
TinyPilot log dump
https://tinypilotkvm.com
Timestamp: 2021-04-21T11:35:42+08:00

Software versions
TinyPilot version: 1.4.1-23-g3322c22 3322c22
uStreamer version: v3.23 74bf710
OS version: Linux up 5.8.0-49-generic #55~20.04.1-Ubuntu SMP Fri Mar 26 01:01:07 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

voltage logs

TinyPilot configuration
[Unit]
Description=TinyPilot - RPi-based virtual KVM
After=syslog.target network.target
StartLimitIntervalSec=0

[Service]
Type=simple
User=tinypilot
WorkingDirectory=/opt/tinypilot
ExecStart=/opt/tinypilot/venv/bin/python app/main.py
Environment=HOST=127.0.0.1
Environment=PORT=8000
Environment=KEYBOARD_PATH=/dev/hidg0
Environment=MOUSE_PATH=/dev/hidg1
Restart=always

[Install]
WantedBy=multi-user.target

TinyPilot logs
Mar 15 17:41:23 up python[860]: [2021-03-15 17:41:23,887] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:42:53 up python[860]: [2021-03-15 17:42:53,962] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:43:23 up python[860]: [2021-03-15 17:43:23,722] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:43:57 up python[860]: [2021-03-15 17:43:57,730] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:45:25 up python[860]: [2021-03-15 17:45:25,838] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:47:27 up python[860]: [2021-03-15 17:47:27,254] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:47:36 up python[860]: [2021-03-15 17:47:36,515] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:47:39 up python[860]: [2021-03-15 17:47:39,717] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:47:43 up python[860]: [2021-03-15 17:47:43,052] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:47:43 up python[860]: [2021-03-15 17:47:43,646] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:48:49 up python[860]: [2021-03-15 17:48:49,575] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:48:50 up python[860]: [2021-03-15 17:48:50,396] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:48:51 up python[860]: [2021-03-15 17:48:51,236] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:48:56 up python[860]: [2021-03-15 17:48:56,715] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:48:57 up python[860]: [2021-03-15 17:48:57,237] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:48:57 up python[860]: [2021-03-15 17:48:57,749] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:49:00 up python[860]: [2021-03-15 17:49:00,595] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:50:31 up python[860]: [2021-03-15 17:50:31,408] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:51:37 up python[860]: [2021-03-15 17:51:37,202] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:51:38 up python[860]: [2021-03-15 17:51:38,353] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:54:05 up python[860]: [2021-03-15 17:54:05,895] INFO in socket_api: Client disconnected
Mar 15 17:54:06 up python[860]: [2021-03-15 17:54:06,062] INFO in socket_api: Client connected
Mar 15 17:54:07 up python[860]: [2021-03-15 17:54:07,073] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:54:07 up python[860]: [2021-03-15 17:54:07,587] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:54:08 up python[860]: [2021-03-15 17:54:08,112] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
Mar 15 17:54:11 up python[860]: [2021-03-15 17:54:11,748] ERROR in socket_api: Failed to forward mouse event: Failed to write to HID interface: /dev/hidg1. Is USB cable connected?
[... cut ...]
	http-request set-var(txn.txnhost) hdr(host)
	use_backend up-tinypilot_ipvANY  if  kvm lanOnly aclcrt_https

frontend http_to_https
	bind			192.168.7.1:10080 name 192.168.7.1:10080   
	mode			http
	log			global
	option			http-keep-alive
	timeout client		30000
	http-request redirect scheme https 

backend up-tinypilot_ipvANY
	mode			http
	id			110
	log			global
	timeout connect		30000
	timeout server		30000
	retries			3
	option			httpchk GET / 
	server			up 192.168.7.2:8888 id 101 check inter 1000  
```

mtlynch, 2021-04-21:
It's hard to diagnose this since it seems like you've got a lot of custom configuration in your network setup.

I believe the root cause is that your HAProxy config isn't forwarding HTTP headers properly to the TinyPilot nginx instance, so the socket.io listener is rejecting the connection because it looks like a CSRF attack (where the `Host` header doesn't match the `Origin` header).

You'll need to watch the HTTP traffic between HAProxy and nginx to figure out what headers HAProxy is sending to nginx. You can do this with ncat. Here's an example command that creates a new ncat proxy on port 8080 and forwards the traffic directly to the TinyPilot instance at port 8000:

```bash
ncat -lkv localhost 8080 -c 'tee /dev/stderr | ncat -v localhost 8000 | tee /dev/stderr'
```

There's [a bug in Flask-SocketIO](https://github.com/miguelgrinberg/Flask-SocketIO/issues/1501) that makes it a little trickier because Flask-SocketIO doesn't process all proxy forwarding headers correctly. The maintainer just fixed it, but it'll probably be a few weeks before the fix is in a published package.

If you don't want to debug the HTTP traffic, a hacky workaround is to change [this line](https://github.com/mtlynch/tinypilot/blob/e953ac014f4b1a341bea784a8fcb8f6197b6fa96/app/socket_api.py#L22) to this:

```python
socketio = flask_socketio.SocketIO(cors_allowed_origins="*")
```

But you'll have to reapply the workaround after every update. It also exposes you to CSRF attacks, which you may or may not consider a credible risk.

lamarios, 2021-04-22:
Thanks for the help, I went with the easy solution. instead of a reverse proxy i just set a redirect to my LAN IP as it is already set up to only allow requests coming from LAN.

mtlynch, 2021-04-22:
Cool, glad to hear you got it working!

closed 2021-04-22 by mtlynch