tiny-pilot/tinypilot #669: Include shortcut to send "CTRL + ALT + DEL" keys to remote device as an Action menu item

demdante, 2021-04-27:
The above feature is usually found as a feature in hypervisor software: allowing the "CTRL + ALT+ DEL" key combination to be sent to the guest virtual machine.  I was thinking that a similar feature would be useful in Tiny Pilot, as the only way that currently exists to have tiny pilot send "CTRL + ALT+ DEL" to the remote machine is to manually press the keys using the virtual keyboard (at least as far as I know).

mtlynch, 2021-04-27:
Thanks for reporting this!

You're correct that the only way to send Ctrl+Alt+Del now is to physically click the buttons on the virtual keyboard. Shortcut macros are definitely on the roadmap.

I'm tracking this in #481, so I'm going to close this one in favor of the original

mtlynch, 2021-06-09:
Reopening, as this bug is smaller in scope than #481, so we can start with this and add more macros / customizability later.

garyjuang, 2021-06-21:
sorry about my English.
how to do reset my remote device before 1.5.3 coming...
any way..like ssh command to tinypilot else...
Thank you

mtlynch, 2021-06-21:
@garyjuang - you can press the ctrl + alt + del buttons on the on-screen keyboard in the meantime

https://user-images.githubusercontent.com/7783288/122758796-3655f700-d267-11eb-8f4f-e2e619835c3d.mp4

garyjuang, 2021-06-21:
@mtlynch 

> @garyjuang - you can press the ctrl + alt + del buttons on the on-screen keyboard in the meantime
> 
>  tinypilot-ctrl-alt-del.mp4

thank you for reply !

But my version is 1.5.1
look like this ...i cant do that like you...

[image]

mtlynch, 2021-06-21:
The video is from an old version of TinyPilot. You can do the same thing in 1.5.1. Scroll down in the browser window so that the Ctrl + Alt keys are visible in the on-screen keyboard and then you can press them + the "Del" key.

garyjuang, 2021-06-22:
> The video is from an old version of TinyPilot. You can do the same thing in 1.5.1. Scroll down in the browser window so that the Ctrl + Alt keys are visible in the on-screen keyboard and then you can press them + the "Del" key.

this is my doing video , but not working....
https://user-images.githubusercontent.com/31755119/122849625-720fc000-d33e-11eb-84f5-04246d23d996.mp4

mtlynch, 2021-06-22:
@garyjuang It looks like you might not be clicking the buttons on the on-screen keyboard. Here's what I recommend:

1. Scroll down to the on-screen keyboard
2. Click the Ctrl button with your mouse
3. Click the Alt button with your mouse
4. Click the Del button with your mouse

https://user-images.githubusercontent.com/7783288/122854938-74dcc780-d2e2-11eb-8584-b96190a8c240.mp4

garyjuang, 2021-06-22:
> @garyjuang It looks like you might not be clicking the buttons on the on-screen keyboard. Here's what I recommend:
> 
> 1. Scroll down to the on-screen keyboard
> 2. Click the Ctrl button with your mouse
> 3. Click the Alt button with your mouse
> 4. Click the Del button with your mouse
> 
>  tinypilot-ctrl-alt-del2.mp4

Thank you for your patient guidance! it works...
sorry my bad habit  must the same time press ctrl + alt + delete

mtlynch, 2021-08-25:
I'd like to resolve this by adding a menu item under Actions that's just labeled "Ctrl + Alt + Del." When the user clicks that menu item, it should send LeftCtrl + RightAlt + Del to the target machine.