tiny-pilot/tinypilot #670: Tiny Pilot display of remote machine freaks out when a particular application window is active

demdante, 2021-04-27:
## Description

Its hard to describe exactly the what I am seeing when the tiny pilot display freaks out, so please see the gif under the screen shot section to see for yourself.  In any event, the issue is that the tiny pilot remote display does not display correctly and kind of freaks out, but only when a particular application window ([Snappy Driver](https://sdi-tool.org/)) is the active window.

## What's the behavior that you expect?

For the tiny pilot remote display to consistently display the remote machine's screen regardless of the active application window.

## What's happening instead?

See Screenshot below

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Launch "Snappy driver" application on remote machine.

## Screenshots

[image]

## Logs

https://logs.tinypilotkvm.com/VmLyDwaz

mtlynch, 2021-04-28:
That's strange. I've never seen that before.

It looks like the screen is flickering a bit before you open Snappy Driver. Is that just an artifact of the GIF capture, or does it always flicker? Do you know if that happens if the machine is hooked up to a physical monitor?

Is it possibly a bad HDMI cable? I've had bizarre capture effects that turned out to be the fault of a bad HDMI cable.

demdante, 2021-05-01:
@mtlynch .   Just to make sure we are on the same page, the issue with the crazy video doesn't occur when snappy driver is launched, but when the window is active.  In the gif that I attached, snappy driver is already running on the system, but minimized.  I simply restore the window, which is when the crazy video occurs.  In answer to your other question, Yes,  The screen flickering that you are seeing prior to the display of the snappy driver window is a product of recording the gif.  The flickering that you are seeing prior to the restoration of the snappy driver window did not actually occur.  I noticed that it seemed like the icons were kind of flickering when I was reviewing the gif after I had captured it, and briefly considered that I should use some alternative means of capturing the issue, but decided to include it, as it accurately shows the main issue that this submission is about: the huge video distortions that occur when the snappy driver window is active.  Also, no.  Prior to using tiny pilot to remotely control the device, I **did** have the device hooked into a physical monitor using the same hdmi cable that is currently connecting the device with  tinypilot.  This issue did not occur then.  Also, I do want to highlight that I have been using tinypilot for about 1-2 months now, and this is the only instance when I experience this phenomena.  Tiny pilot works great except in this specific circumstance.  That seems to indicate that it is not a hardware issue.  I would think the issue would be far more.,..random...if that was the case.  I will try switching the hdmi cable to cover all the bases, but I doubt that will resolve the issue.  I will get back to you once I have switched the cable.

mtlynch, 2021-06-08:
@demdante any updates here? Are you still seeing this with a known-good HDMI cable?

heurekus, 2021-06-12:
It's the neon-yellow color in the window that produces this! I can reproduce this with a 2109 USB video capture device!

Have a look at the following two screenshots I took: 

Image 1) An Outlook window open with black/white text/background. All fine.

Image 2) Then I scroll down to text that is yellow. And boom, screen below the yellow goes all crazy and starts to flicker. It only happens if the yellow part is of a certain width. 

Image 3) For only a few characters in yellow everything is fine. No problem here.

The flickering goes away as soon as the neon-yellow color goes away.

How strange!?

[image]

[image]

[image]

heurekus, 2021-06-14:
Update: **I can get rid of the flickering** by changing the fps from 30 to 20. 

Info: I tried with a different cable, different notebook and different OS on the device (Ubuntu instead of Windows) and I get the same flickering (+fix).

mtlynch, 2021-06-14:
Wow, that's really strange! I'll reach out to the maintainer of the video streaming library to see if they have any ideas about why that's happening with that particular color.