tiny-pilot/tinypilot #677: Add "Reset Defaults" button in Video Settings dialog

mtlynch, 2021-05-05:
If the user changes their video settings, there's no way to restore defaults unless they know what the defaults are.

[image]

We should add a "Restore Defaults" button that resets the settings to their original values.

### Implementation suggestion

I think a good way to implement this would be that the dialog sends `DELETE` requests for each setting and then pulls down the settings again via `GET`.

jotaen4tinypilot, 2021-05-13:
Would the “reset” button apply the changes right away and close the dialog? If not (i.e. the “reset” should only bring the sliders to their original position): would there be one “reset” per slider or one overall “reset” for all of them?

Also, we could consider making the sliders a bit wider. There is plenty of space to the sides, and it would allow the user to adjust the setting more fine-granularly.

mtlynch, 2021-05-13:
>Would the “reset” button apply the changes right away and close the dialog? If not (i.e. the “reset” should only bring the sliders to their original position): 

Oh, that's a good point. If we delete the settings on the server side, resetting would automatically apply the settings.

Maybe instead of `DELETE` we can add another route like `/settings/video/fps/default` (GET) so the frontend can just reset the sliders back to their original values, but the user can still cancel

>would there be one “reset” per slider or one overall “reset” for all of them?

Let's do just one reset for everything, for the sake of simplicity.

>Also, we could consider making the sliders a bit wider. There is plenty of space to the sides, and it would allow the user to adjust the setting more fine-granularly. (Not sure it’s worth opening a separate issue?)

That's a good idea. I've just tweaked it a bit in #691

PR #705 (2021-05-26): Add "Reset Defaults" button in Video Settings dialog
[Video demo](https://www.loom.com/share/5d79b2e867ba41c7a99e039b2a4d4d89).

Fixes #677
closed 2021-05-26 by jdeanwallace