tiny-pilot/tinypilot #683: Power/Data splitter is there plans for a USB C power port version?

oiaohm, 2021-05-08:
I really find the current design of the Tiny Pilot power/data splitter annoying because it so close to perfect but its just not quite.   Like I have proper raspberry pi 4 and clone raspberry pi 4 power supplies.   They are USB C plugged non removable cables from brick having to use some form of adaptor to take them down to usb micro to use the splitter seams stupid and just adding to the mess of cables.    Yes having the power going into the splitter USB C and data as micro and the output as USB A would be what you call Sod law protection.   Currently with the two micro usb I see the possibility of screwing up the wiring a lot more.

I have not found plans to the Tiny Pilot power/data splitter circuit.

Also for the production unit tinypilot Voyager  I would suggest looking at the argon one if doing a revision for way of dealing with side ports.   The argon one has a board down the side of the pi that convert the two small Hdmi to large hdmi while rerouting all the ports to one side of the device.(the argon one powers the pi by the gpio header)   In theory that board could be made longer to cover the power in and data lines to bring all ports to setup to one side of the device.  I guess the board down the side would have enough room for the protection circuit.   Less of a wiring nightmare.   Cables going off at 90 degrees to each other in a rack can cause major pain by increase risk of getting hooked.

Yes in a lot of ways I would take side board expanding the ports and exposing otg port over the spitter box.   This is all due to my personal dislike for extra cables in rack.   Yes I would class doing a full side board as a possible new Power/Data spitter.

mtlynch, 2021-05-10:
Thanks for reaching out!

I fully agree about the inconvenience of the extra splitter box. I'm working on a custom HAT that will provide PoE or direct microUSB power input without the need for the goofy extra box. I was hoping to have it by now, but the global chip shortage has thrown a bit of a wrench in the gears, with components I need selling out before I can even design a prototype.

I'm still moving forward on this, and I'm aiming to have an all-in-one unit by mid-summer, and a [rack-mountable version](https://tinypilotkvm.com/blog/rackmount) by end of summer.

closed 2021-05-10 by mtlynch

oiaohm, 2021-05-13:
On Tue, May 11, 2021 at 5:29 AM Michael Lynch ***@***.***>
wrote:

> Thanks for reaching out!
>
> I fully agree about the inconvenience of the extra splitter box. I'm
> working on a custom HAT that will provide PoE or direct microUSB power
> input without the need for the goofy extra box. I was hoping to have it by
> now, but the global chip shortage has thrown a bit of a wrench in the
> gears, with components I need selling out before I can even design a
> prototype.
>
> I'm still moving forward on this, and I'm aiming to have an all-in-one
> unit by mid-summer, and a rack-mountable version
> <https://tinypilotkvm.com/blog/rackmount> by end of summer.
>
>
> I am not just looking at the ip-kvm usage case.     Think doing a stream
deck with a raspberry pi touch screen.   So able to custom macro key all
the keyboard and input devices plugged into pi to another computer.   Or
the raspberry pi being display
https://www.phoronix.com/scan.php?page=news_item&px=Generic-USB-Display-GUD-5.13
Yes GUD the Generic USB Display option.

I see a place for prototyping for some users the spitter box will be
good.    Do note the stream deck usage or the GUD case you would not be
needing the PoE power but you may be wanting external power for stability
of course..

microUSB power input << That a problem.  Take a note of what I said I have
proper raspberry pi 4 power supplies.  So what I am after there is proper
USB C power input so my existing power supplies for raspberry pi 4 plug
straight in.

A side board instead of a hat to bring all the side ports to one side would
be a good thing.   Do look at using vertical full size hdmi for the display
outputs and inputs.  Turning the hdmi ports vertical you can fit 2 in the
space of one.
When I say vertical I mean the following connectors
https://au.element14.com/te-connectivity/2007435-1/receptacle-hdmi-flag-tht/dp/1816577
.    Next you still do need hdmi out with the rack-mountable version.   Do
you really want to have to alter cables around when you are at the rack to
get an in person screen/keyboard connected?.     Next in the rack mountable
I don't understand the fan position? Why is it to the side on the front.
These commonly are at the back of these devices with venting in the front
to draw air over the heat sinks.

If you are having a hat I would have put the micro sd above the raspberry
pi main board if possible to cut down on width.   Question is also how
often the micro sd going to need being replaced.

Do think about width and how many you could place side by into a single 19
inch U.   Its very wasteful of rack space to have a small single item in a
1U.   Slightly shorter than a single U is common so that when you take the
side off the item can placed in
https://www.selby.com.au/brands/raxx/1u-19x15in-aluminium-vented-shelf-for-rack-system-black.html
like shelf.  Yes zip tied down.   Also some small items that open up the
top of case bolt join them to each other to make a complete 1U unit.

https://www.amazon.com.au/Raspberry-Rack-Mount-inch-Units/dp/B085LQT67P

If this is anything to go by you should be able to get at least 3 units to
possible 4 units in a single 19 inch U.   Of course this requires careful
control of space usage.   You don't have that much room on the front
really.  Only really 2 hdmi ports worth extra to the in width.  Stacking
ports will be important.   Yes power not by ethernet is in rack devices is
commonly on the back.

Next problem to consider is most rack cases ATX the keyboard and monitor
ports are on the back.   It is really uncommon to have a front keyboard and
monitor ports.
https://www.aten.com/au/en/products/kvm/rack-kvm-switches/cs18208/
even your common rack KVM switchs have control ports on the back.    So
HDMI in and keyboard out on the front does not make much sense once you are
using it in a rack.    Now even with the HDMI out there is a good question
if one of those should be on the back and if one of the front USB ports of
the pi should be routed to the back as well.

I do understand that the back ports on KVM switch have the habit of being
keyboard and mouse ports only so you want to plug into the front ports with
the USB so you can emulate them but the HDMI port is going to be on the
back.   Do also notice the USB type B plugs for the cables coming from the
computer to the KVM.  This is sod law prevention.

It is more common in audio racks to have the hdmi and keyboard ports at the
front than the back but the back is still common 19 inch audio racks as
well for the HDMI and keyboard ports to be at the back.  General server
racks are almost 100 percent that the keyboard and monitor ports will be at
the back.   So having the input ports at the front are going to cause cable
routing nightmares.

There are a handful of ip-kvm device with a HDMI port out and keyboard on
the front for you to plug a portable HDMI and keyboard in for
diagnostics.   Basically HDMI in you don't find on the front in common rack
server setups

I hope you don't take offence by all the possible issues with the rack
mountable design I spotted.     I did do server administration and
installation for over 15 years of rack mounted servers..   Sod law
prevention is important, same with thinking where the cables will naturally
be. You don't want to have cables crossing from the front to the back of
the rack any more than you have to.

Peter Dolding

mtlynch, 2021-05-25:
Thanks, Peter!

I agree with your feedback. I'm going to redesign the rack version so that the HDMI capture and emulated keyboard/mouse USB port is in the back.

We wouldn't gain much by putting in two HDMI ports since the Pi has only one CSI bridge, so we're limited to one HDMI input per Pi. I'm not exposing the Pi's HDMI outputs in the initial version because there's not enough utility from them.

I agree about better using the space. I'm going to try to make the mount modular so you can fit 1-3 TinyPilots inside.