tiny-pilot/tinypilot #694: Erratic keyboard behaviour on MacOS (Meta key?)

jotaen, 2021-05-14:
I noticed that the Meta key can get stuck, so at times it isn’t released properly for whatever reason.

This happened while checking out the Virtual Media compability problems. I’m opening this ticket as note to self for now, will add more info when investigating further.

mtlynch, 2021-05-14:
Oh, interesting. We should try to repro this after #692 is merged in to see if that makes any difference.

jotaen, 2021-05-21:
This seems to be resolved through the fix in #692.

I also think I know what the issue was or how to reproduce it:

1. Open TinyPilot web interface
2. Press ⌘ (Meta) + Tab in order to switch to a different app on your host system
3. Keydown for ⌘ (Meta) was send to remote machine, but not the corresponding keyup and also not the tab, because at the time both keys are released the browser is not in the foreground anymore, so the TinyPilot web app never receives these events.
4. ⌘ (Meta) is now pressed all the time on the remote machine. Even if you press ⌘ (Meta) again it wouldn’t release it, only pressing `ESC` does.

Anyway, by running the fix this doesn’t occur anymore.

closed 2021-05-21 by jotaen

mtlynch, 2021-05-21:
Thanks for verifying!

>⌘ (Meta) is now pressed all the time on the remote machine. Even if you press ⌘ (Meta) again it wouldn’t release it, only pressing ESC does. (At least if the remote system is a Mac.)

Hmm, this doesn't seem like *great* behavior. We'll see how users find the new implementation. We might have to revisit this and find better ways of handling Meta so that it doesn't get stuck. I think the behavior's the same on all OSes because meta changes focus from the current window.