tiny-pilot/tinypilot #695: Bad Gateway from video stream

DrRek, 2021-05-16:
## Description

Tiny pilot is capturing my mouse&keyboard but is not showing the screen.

## What's the behavior that you expect?

I should be able to see the screen

## What's happening instead?

I should be able to see the screen

## What are the steps to reproduce this behavior?

1. use a raspberry pi zero and a CSI to HDMI adapter
2. run```
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```
3. screen says "no screen" (or smth similar)
4. after googling realize I should run the following steps
```
 cat /home/tinypilot/settings.yml
ustreamer_port: 8001
ustreamer_persistent: true
ustreamer_encoder: hw
ustreamer_format: jpeg
ustreamer_resolution: 1920x1080
ustreamer_capture_device: "tc358743"
```
4. run `sudo /opt/tinyplot-privileged/update && sudo reboot`
5. still no screen (as shown in the screenshot below)

## Screenshots

![image](https://user-images.githubusercontent.com/12783153/118400308-5d664b00-b661-11eb-838f-4bbc03db10a7.png)

## Logs

 https://logs.tinypilotkvm.com/xKCICJrb

mtlynch, 2021-05-17:
The normal installer assumes that you have an HDMI to USB dongle. I should fix those instructions.

You should be able to fix it with these commands:

```bash
echo 'ustreamer_port: 8001
ustreamer_persistent: true
ustreamer_capture_device: "tc358743"' > /home/tinypilot/settings.yml
sudo /opt/tinyplot-privileged/update && sudo reboot
```

closed 2021-05-17 by mtlynch

DrRek, 2021-05-17:
I already had `ustreamer_capture_device: "tc358743"` in my config.
However, I followed your steps but still no results :(

[image]

 https://logs.tinypilotkvm.com/SpkoAKis

DrRek, 2021-05-17:
Fixed, there were some typos in the command you mentioned.

posted for future reference:
```
echo 'ustreamer_port: 8001
ustreamer_persistent: true
ustreamer_capture_device: tc358743' > /home/tinypilot/settings.yml
sudo /opt/tinypilot-privileged/update && sudo reboot
```

mtlynch, 2021-05-18:
Glad you got it working!

>Fixed, there were some typos in the command you mentioned.

>posted for future reference:

The only difference is the quotes around `tc358743`? That's strange because that shouldn't make a difference.

DrRek, 2021-05-23:
the qoute in the command (which does make a difference in the sh scritp) and the typo "tinypilot" instead of "tinyplot" :)

mtlynch, 2021-05-24:
Ah, thanks!