tiny-pilot/tinypilot #704: Failed command (python3) ensurepip is not available on upgrade.

dharrigan, 2021-05-21:
## Description

Upgrade script fails

## What's the behavior that you expect?

Upgrade script succeeds

## What's happening instead?

```
+ pushd /opt/tinypilot-updater
/opt/tinypilot-updater ~
+ python3 -m venv venv
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/opt/tinypilot-updater/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']
```

## What are the steps to reproduce this behavior?

1. ssh into raspberrypi
2. sudo to root
3. run `/opt/tinypilot/scripts/upgrade`

## Logs

```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

root@raspberrypi:~# cat /etc/issue
Raspbian GNU/Linux 10 \n \l

root@raspberrypi:~# /opt/tinypilot/scripts/upgrade 
+ set -u
+ set -e
+ curl --silent --show-error https://raw.githubusercontent.com/mtlynch/tinypilot/master/quick-install
+ bash -
+ readonly DEFAULT_TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ DEFAULT_TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ [[ -n '' ]]
+ set -u
++ mktemp --tmpdir=/var/tmp --suffix .yml
+ readonly TINYPILOT_SETTINGS_FILE=/var/tmp/tmp.iBLBLtHvm3.yml
+ TINYPILOT_SETTINGS_FILE=/var/tmp/tmp.iBLBLtHvm3.yml
+ [[ -f /home/tinypilot/settings.yml ]]
+ echo 'Using settings file at: /home/tinypilot/settings.yml'
Using settings file at: /home/tinypilot/settings.yml
+ sudo cp /home/tinypilot/settings.yml /var/tmp/tmp.iBLBLtHvm3.yml
+ readonly EXTRA_VARS_PATH=@/var/tmp/tmp.iBLBLtHvm3.yml
+ EXTRA_VARS_PATH=@/var/tmp/tmp.iBLBLtHvm3.yml
+ add_setting_if_undefined ustreamer_port 8001
+ local key=ustreamer_port
+ local value=8001
+ grep --silent '^ustreamer_port:' /var/tmp/tmp.iBLBLtHvm3.yml
+ add_setting_if_undefined ustreamer_persistent true
+ local key=ustreamer_persistent
+ local value=true
+ grep --silent '^ustreamer_persistent:' /var/tmp/tmp.iBLBLtHvm3.yml
+ USE_TC358743_DEFAULTS=
+ grep --silent '^ustreamer_capture_device:' /var/tmp/tmp.iBLBLtHvm3.yml
+ '[' -f /home/ustreamer/config.yml ']'
+ grep --silent 'capture_device: "tc358743"' /home/ustreamer/config.yml
+ [[ '' == \y ]]
+ add_setting_if_undefined ustreamer_encoder hw
+ local key=ustreamer_encoder
+ local value=hw
+ grep --silent '^ustreamer_encoder:' /var/tmp/tmp.iBLBLtHvm3.yml
+ add_setting_if_undefined ustreamer_format jpeg
+ local key=ustreamer_format
+ local value=jpeg
+ grep --silent '^ustreamer_format:' /var/tmp/tmp.iBLBLtHvm3.yml
+ add_setting_if_undefined ustreamer_resolution 1920x1080
+ local key=ustreamer_resolution
+ local value=1920x1080
+ grep --silent '^ustreamer_resolution:' /var/tmp/tmp.iBLBLtHvm3.yml
+ echo 'Final install settings:'
Final install settings:
+ cat /var/tmp/tmp.iBLBLtHvm3.yml
ustreamer_encoder: hw
ustreamer_format: jpeg
ustreamer_persistent: true
ustreamer_port: 8001
ustreamer_quality: 50
ustreamer_resolution: 1920x1080
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ '[' . = . ']'
+ SCRIPT_DIR=/opt/tinypilot
+ TINYPILOT_README=/opt/tinypilot/README.md
+ '[' -f /opt/tinypilot/README.md ']'
++ head -n 1 /opt/tinypilot/README.md
+ '[' '# TinyPilot' = '# TinyPilot Pro' ']'
+ '[' 0 = 1 ']'
+ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
+ sudo apt-get install -y git libffi-dev libssl-dev python3-dev python3-venv
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u3).
libffi-dev is already the newest version (3.2.1-9).
python3-dev is already the newest version (3.7.3-1).
python3-venv is already the newest version (3.7.3-1).
libssl-dev is already the newest version (1.1.1d-0+deb10u6+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ INSTALLER_DIR=/opt/tinypilot-updater
+ sudo mkdir -p /opt/tinypilot-updater
++ whoami
++ whoami
+ sudo chown root:root --recursive /opt/tinypilot-updater
+ pushd /opt/tinypilot-updater
/opt/tinypilot-updater ~
+ python3 -m venv venv
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/opt/tinypilot-updater/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']
```

python3-venv is already installed:

```
root@raspberrypi:~# apt policy python3-venv
python3-venv:
  Installed: 3.7.3-1
  Candidate: 3.7.3-1
  Version table:
 *** 3.7.3-1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
        100 /var/lib/dpkg/status
```

and LANG is set:

```
root@raspberrypi:~# echo $LANG
en_GB.UTF-8
```

mtlynch, 2021-05-24:
Thanks for reporting this! Does this reproduce consistently on a clean Raspbian install?

dharrigan, 2021-05-25:
Hi,

From a fresh clean install, it appears to work. I executed the upgrade script twice, with a reboot inbetween.

I will close the ticket.

-=david=-

closed 2021-05-25 by dharrigan