tiny-pilot/tinypilot #717: Properly disable and renable tinypilot

DrRek, 2021-05-31:
Hello,

I've tinypilot installed on a raspberry that I also use for many other things.
I only use tinypilot when some upgrade breaks the system and I have to access to remote to my machine (less than once per month).
Therefore I would like to lighten the load on the raspberry while I don't use it and launch tinypilot on demand while needed.
I tried stopping the tinypilot service however I noticed that there are still some "ustreamer" processes running.

Is there a proper way to properly shutdown/restart tinypilot functionalities?

DrRek, 2021-05-31:
also nginix keeps running

mtlynch, 2021-06-01:
TinyPilot doesn't really use much in terms of system resources, especially when you're not viewing the web interface. But the full list of TinyPilot's systemd services is:

* `tinypilot`
* `ustreamer`
* `usb-gadget`
* `nginx`

You should be able to stop and disable them through the normal systemd CLI commands.

closed 2021-06-01 by mtlynch