tiny-pilot/tinypilot #732: Virtual Keyboard: don’t process to key events

jotaen, 2021-06-23:
After you click a random key (say “A”) on the virtual keyboard and then press “space” or “return” on your physical keyboard, it sends the following sequence: `Space`, `A`. That’s because the “A”’s `<div>` element still has focus, so the space or return key makes the virtual key emit the current key another time. Only after you click somewhere else, the `<div>` looses focus and this behaviour stops.

demdante, 2021-06-25:
I think the issue that I am experiencing is related to this issue as well.  My issue involves the "space" key on the virtual keyboard just not processing any of the space key events.  The space key on the virtual keyboard just....does not work....  You can see what I mean by viewing the video at the below link:

https://www.dropbox.com/s/c82dqycgytn99cp/20210624_221129.mp4?dl=0

jotaen, 2021-06-25:
@demdante the issue you’ve been experiencing was actually [a separate one](https://github.com/tiny-pilot/tinypilot/issues/729), but we [pushed a fix for that just now](https://github.com/tiny-pilot/tinypilot/pull/736). So your space key should work again after you update your device (“System” → “Update”).