tiny-pilot/tinypilot #751: Keyboard and Mouse Input Not Transmitted

lensherm, 2021-07-23:
## Description

On a fresh install of RPi OS, the keyboard and mouse input isn't being transmitted to host

## What are the steps to reproduce this behavior?

## Logs

TinyPilot log dump
https://tinypilotkvm.com
Timestamp: 2021-07-23T10:45:15-04:00

Software versions
TinyPilot version: 1.5.2-2-gfcfbd1b fcfbd1b
uStreamer version: v3.26 a97f08e
OS version: Linux TinyPilot 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux

TinyPilot state
temp=35.0'C
throttled=0x0

voltage logs

TinyPilot configuration
[Unit]
Description=TinyPilot - RPi-based virtual KVM
After=syslog.target network.target
StartLimitIntervalSec=0

[Service]
Type=simple
User=tinypilot
WorkingDirectory=/opt/tinypilot
ExecStart=/opt/tinypilot/venv/bin/python app/main.py
Environment=HOST=127.0.0.1
Environment=PORT=8000
Environment=KEYBOARD_PATH=/dev/hidg0
Environment=MOUSE_PATH=/dev/hidg1
Restart=always

[Install]
WantedBy=multi-user.target

TinyPilot logs
Jul 23 10:45:09 TinyPilot python[582]:     self._target(*self._args, **self._kwargs)
Jul 23 10:45:09 TinyPilot python[582]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
Jul 23 10:45:09 TinyPilot python[582]:     hid_handle.write(bytearray(buffer))
Jul 23 10:45:09 TinyPilot python[582]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Jul 23 10:45:10 TinyPilot python[582]: Process Process-3:
Jul 23 10:45:10 TinyPilot python[582]: Traceback (most recent call last):
Jul 23 10:45:10 TinyPilot python[582]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jul 23 10:45:10 TinyPilot python[582]:     self.run()
Jul 23 10:45:10 TinyPilot python[582]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jul 23 10:45:10 TinyPilot python[582]:     self._target(*self._args, **self._kwargs)
Jul 23 10:45:10 TinyPilot python[582]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
Jul 23 10:45:10 TinyPilot python[582]:     hid_handle.write(bytearray(buffer))
Jul 23 10:45:10 TinyPilot python[582]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Jul 23 10:45:10 TinyPilot python[582]: Process Process-4:
Jul 23 10:45:10 TinyPilot python[582]: Traceback (most recent call last):
Jul 23 10:45:10 TinyPilot python[582]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jul 23 10:45:10 TinyPilot python[582]:     self.run()
Jul 23 10:45:10 TinyPilot python[582]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 99, in run
Jul 23 10:45:10 TinyPilot python[582]:     self._target(*self._args, **self._kwargs)
Jul 23 10:45:10 TinyPilot python[582]:   File "/opt/tinypilot/app/hid/write.py", line 18, in _write_to_hid_interface_immediately
Jul 23 10:45:10 TinyPilot python[582]:     hid_handle.write(bytearray(buffer))
Jul 23 10:45:10 TinyPilot python[582]: BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
Jul 23 10:45:10 TinyPilot python[582]: Process Process-5:
Jul 23 10:45:10 TinyPilot python[582]: Traceback (most recent call last):
Jul 23 10:45:10 TinyPilot python[582]:   File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
Jul 23 10:45:10 TinyPilot python[582]:     self.run()
[... 247 lines cut ...]
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /third-party/underscore.js/1.11.0/underscore-min.js HTTP/1.1" 200 18618 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /third-party/fonts/overpass-light.woff2 HTTP/1.1" 200 26568 "http://tinypilot.local/third-party/fonts/fonts.css" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /third-party/fonts/overpass-semibold.woff2 HTTP/1.1" 200 25032 "http://tinypilot.local/third-party/fonts/fonts.css" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /js/settings.js HTTP/1.1" 200 1194 "http://tinypilot.local/js/app.js" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJnHrs&sid=mqkbQ9I1ywkGvm2oAAAC HTTP/1.1" 200 1 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJnHtX&sid=mqkbQ9I1ywkGvm2oAAAC HTTP/1.1" 200 1 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:29:08 -0400] "GET /favicon.ico HTTP/1.1" 304 0 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:34:16 -0400] "GET /socket.io/?EIO=4&transport=websocket&sid=mqkbQ9I1ywkGvm2oAAAC HTTP/1.1" 101 735 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:34:16 -0400] "GET /stream?advance_headers=1 HTTP/1.1" 200 566732855 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:34:38 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJoahj HTTP/1.1" 502 575 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:34:43 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJodIt HTTP/1.1" 200 96 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:34:43 -0400] "POST /socket.io/?EIO=4&transport=polling&t=NhJodKL&sid=CoySmkQB5R6wzsgQAAAA HTTP/1.1" 200 2 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:34:43 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJodKN&sid=CoySmkQB5R6wzsgQAAAA HTTP/1.1" 200 1 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:35:28 -0400] "GET /socket.io/?EIO=4&transport=websocket&sid=CoySmkQB5R6wzsgQAAAA HTTP/1.1" 101 83 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:37:15 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJp8hr HTTP/1.1" 200 96 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:37:15 -0400] "POST /socket.io/?EIO=4&transport=polling&t=NhJp8la&sid=3fQFu06fWgxS_wThAAAC HTTP/1.1" 200 2 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:37:15 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJp8lb&sid=3fQFu06fWgxS_wThAAAC HTTP/1.1" 200 1 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:37:15 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJp8l-&sid=3fQFu06fWgxS_wThAAAC HTTP/1.1" 200 1 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
[... cut ...]
192.168.1.81 - - [23/Jul/2021:10:42:33 -0400] "GET /socket.io/?EIO=4&transport=polling&t=NhJqMT6&sid=OjtOe0b5XqW-TFw9AAAA HTTP/1.1" 200 1 "http://tinypilot.local/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"
192.168.1.81 - - [23/Jul/2021:10:44:10 -0400] "GET /socket.io/?EIO=4&transport=websocket&sid=OjtOe0b5XqW-TFw9AAAA HTTP/1.1" 101 66 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.164 Safari/537.36" "-"

mtlynch, 2021-07-23:
Thanks for reporting this!

What's the OS / hardware of the target computer? Does it work if you connect the TinyPilot to other computers?

What model Pi are you using and what model USB-C to USB-A cable are you using to connect the Pi to the computer?

lensherm, 2021-07-23:
- So faI've tried connecting to a Late 2012 Mac Mini running Proxmox and to a Dell Inspiron 3880 running Windows 10, which I just pulled out of the box earlier this morning.

- In terms of the Pi, I've only tried TinyPilot with one of my Pi 4 4Gb. I have a couple more and one 8Gb. As for the cable, I am using Anker Powerline+ USB-C to USB 3.0, although I tried a couple off brands, just to be sure. I also tried with and without this thing [here](https://www.pishop.us/product/usb-c-pwr-splitter/).

mtlynch, 2021-07-23:
I can't figure out what could be the problem here.

Usually it's one of these issues:

* USB-C to USB-A cable: unlikely, since the Anker one should work and you've tried several others
* Incompatible target computer: shouldn't be, since you've tried multiple that are known compatible
* Defective Pi: This never happens, but even if it did, you've tried multiple Pis
* Wrong OS: You're using the Raspberry Pi OS image, which is known compatible

Are you connecting directly to the Pi's USB-C port?

The last time I was this stumped, it turned out that the user was connecting through the Argon ONE case, which intercepts the USB-C port in an incompatible way.

lensherm, 2021-07-24:
BINGO!!!!

Argon ONE strikes again. All is well in the world as soon as I connected directly!

Thank you, Mike.

lensherm, 2021-07-24:
By the way, might want to put a DO NOT USE ARGON ONE everywhere in the documentation :-)

closed 2021-07-24 by lensherm

michaelguild13, 2022-03-21:
@mtlynch I am having a similar issue with a target machine that's a macbook pro 2019 (monterey). I am using usb-c to usb-c for the data.