tiny-pilot/tinypilot #752: HDMI screen issue in Ubuntu 20.04.2 LTS  64-bit

phan-pivotal, 2021-07-26:
## Description

Briefly summarize the problem you're experiencing.

## What's the behavior that you expect?

HDMI screen show normally in Ubuntu 20.04.2 LTS  64-bit.
(Linux ubuntu 5.4.0-1041-raspi #45-Ubuntu SMP PREEMPT Thu Jul 15 01:17:56 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux )

## What's happening instead?

A blank section in the web.

## What are the steps to reproduce this behavior?

1. Open TinyPilot in Chrome
2. Click on '....'
4. See error

## Screenshots

[image]

## Logs

Please paste the URL you see when you run `/opt/tinypilot/dev-scripts/dump-logs`
http://logs.tinypilotkvm.com/U501YihG

mtlynch, 2021-07-26:
Thanks for reporting this!

Unfortunately, TinyPilot is currently only supported on Raspberry Pi OS armv7.

phan-pivotal, 2021-07-31:
In fact, if "apt update" executed on Raspberry Pi OS armv7, similar issues can be observed.
So I guess this is a package incompatible issue.