tiny-pilot/tinypilot #753: ChromeOS support treats mouse input as a touch device

mtlynch, 2021-08-04:
## Description

A user has reported that Chromebook interprets TinyPilot's mouse emulation as a touch device. 

## What's the behavior that you expect?

The ChromeOS treats the TinyPilot's input the same way it would treat an external physical mouse.

## What's happening instead?

Instead of moving the cursor as the user moves the cursor on the screen, the cursor only moves when the user clicks the cursor. And scrolling can only happen when the user clicks and drags the screen like on a phone.

## What are the steps to reproduce this behavior?

1. Connect TinyPilot to a Chromebook
2. Try moving the mouse

## Screenshots

Screenshot is of a long-left click of the word "easy"

[image]

## Logs

https://logs.tinypilotkvm.com/q6Flf5S3

gkovacs, 2023-05-29:
I can confirm that I can reproduce this issue with the Tinypilot Voyager 2a, firmware version 2.5.4, when attached to a HP Chromebook x2 11-DA0023DX. I have not found a workaround.

Belal-Salas, 2024-08-14:
Any update on this?

cghague, 2024-08-15:
We're looking into [relative mouse mode](https://github.com/tiny-pilot/tinypilot/issues/853), which could potentially be the solution. However, we don't have a timeframe for when/if we will implement it, so unfortunately, there isn't an update on this issue.