tiny-pilot/tinypilot #764: `apt-get` update not working

jdeanwallace, 2021-08-24:
## Description

I get an error when running the TinyPilot quick-install script like so:

```bash
curl \
  --silent \
  --show-error \
  https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/quick-install | \
    bash - && \
  sudo reboot
```

## What's the behavior that you expect?

I expect TinyPilot to be installed and the quick-install script to complete successfully.

## What's happening instead?

I get the following output and error:

```bash
$ curl \
>   --silent \
>   --show-error \
>   https://raw.githubusercontent.com/tiny-pilot/tinypilot/master/quick-install | \
>     bash - && \
>   sudo reboot
+ readonly DEFAULT_TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ DEFAULT_TINYPILOT_SETTINGS_FILE=/home/tinypilot/settings.yml
+ [[ -n '' ]]
+ set -u
++ mktemp --tmpdir=/var/tmp --suffix .yml
+ readonly TINYPILOT_SETTINGS_FILE=/var/tmp/tmp.j9nQvT9vag.yml
+ TINYPILOT_SETTINGS_FILE=/var/tmp/tmp.j9nQvT9vag.yml
+ [[ -f /home/tinypilot/settings.yml ]]
+ echo 'Using settings file at: /home/tinypilot/settings.yml'
Using settings file at: /home/tinypilot/settings.yml
+ sudo cp /home/tinypilot/settings.yml /var/tmp/tmp.j9nQvT9vag.yml
+ readonly EXTRA_VARS_PATH=@/var/tmp/tmp.j9nQvT9vag.yml
+ EXTRA_VARS_PATH=@/var/tmp/tmp.j9nQvT9vag.yml
+ add_setting_if_undefined ustreamer_port 8001
+ local key=ustreamer_port
+ local value=8001
+ grep --silent '^ustreamer_port:' /var/tmp/tmp.j9nQvT9vag.yml
+ add_setting_if_undefined ustreamer_persistent true
+ local key=ustreamer_persistent
+ local value=true
+ grep --silent '^ustreamer_persistent:' /var/tmp/tmp.j9nQvT9vag.yml
+ USE_TC358743_DEFAULTS=
+ grep --silent '^ustreamer_capture_device:' /var/tmp/tmp.j9nQvT9vag.yml
+ '[' -f /home/ustreamer/config.yml ']'
+ grep --silent 'capture_device: "tc358743"' /home/ustreamer/config.yml
+ [[ '' == \y ]]
+ add_setting_if_undefined ustreamer_encoder hw
+ local key=ustreamer_encoder
+ local value=hw
+ grep --silent '^ustreamer_encoder:' /var/tmp/tmp.j9nQvT9vag.yml
+ add_setting_if_undefined ustreamer_format jpeg
+ local key=ustreamer_format
+ local value=jpeg
+ grep --silent '^ustreamer_format:' /var/tmp/tmp.j9nQvT9vag.yml
+ add_setting_if_undefined ustreamer_resolution 1920x1080
+ local key=ustreamer_resolution
+ local value=1920x1080
+ grep --silent '^ustreamer_resolution:' /var/tmp/tmp.j9nQvT9vag.yml
+ echo 'Final install settings:'
Final install settings:
+ cat /var/tmp/tmp.j9nQvT9vag.yml
ustreamer_encoder: hw
ustreamer_format: jpeg
ustreamer_persistent: true
ustreamer_port: 8001
ustreamer_resolution: 1920x1080
tinypilot_repo_branch: master
+ HAS_PRO_INSTALLED=0
++ dirname bash
+ SCRIPT_DIR=.
+ '[' . = . ']'
+ SCRIPT_DIR=/opt/tinypilot
+ TINYPILOT_README=/opt/tinypilot/README.md
+ '[' -f /opt/tinypilot/README.md ']'
+ '[' 0 = 1 ']'
+ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
```

It seems like the error is coming from:

```bash
$ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
```

## What are the steps to reproduce this behavior?

1. `ssh tinypilot`
2. `sudo apt-get update`

## Workaround

I managed to get around this issue by running `sudo apt update` instead of `sudo apt-get update`:

```bash
$ sudo apt update -y
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] Y
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.0 MB in 11s (1137 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
183 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

mtlynch, 2021-08-24:
Looks like it's due to this:

https://www.raspberrypi.org/forums/viewtopic.php?t=318302&p=1905442

The man page discourages `apt` in shell scripts because the UI is not guaranteed backwards compatible.

>All features of apt(8) are available in dedicated APT tools like apt-get(8) and apt-cache(8) as well.  apt(8) just changes the default value of some options... So you should prefer using these commands (potentially with some additional options enabled) in your scripts as they keep backward compatibility as much as possible.

It looks like the fix we want is:

```
sudo apt-get update --allow-releaseinfo-change
```

mtlynch, 2021-08-24:
Actually, if there's a way to explicitly specify `buster`, that would be better. I'm a little uncomfortable with telling apt it's allowed to just upgrade major distribution versions whenever it feels like it.

mtlynch, 2021-08-24:
Okay, it looks like this works:

```bash
readonly DEBIAN_RELEASE="buster"
sudo apt-get update --target-release "${DEBIAN_RELEASE}"
```
```

PR #768 (2021-08-24): Explicitly target buster as release version during apt-get update
Fixes #764

---
This change is [[image]](https://reviewable.io/reviews/tiny-pilot/tinypilot/768)
commit 09f56535 (2021-08-24): Explicitly target buster as release version during apt-get update (#768)

Fixes #764
closed 2021-08-24 by mtlynch