tiny-pilot/tinypilot #769: Simplify request parsing and validation

jotaen, 2021-08-24:
The [`request_parsers`](https://github.com/tiny-pilot/tinypilot/tree/master/app/request_parsers) package contains logic that’s mostly concerned with extracting data from a JSON payload. The validation of the data is split off into a [sub-package `validators`](https://github.com/tiny-pilot/tinypilot/tree/master/app/request_parsers/validators). 

- Both the implementation and the tests of the request parsers are quite repetitive, while the parsing logic in itself is rather trivial.
- The fact that we keep parsing and validating separately, “forces” us to write redundant tests. See e.g. the video settings, where the same logic is tested in both the [parser](https://github.com/tiny-pilot/tinypilot/blob/4d0fdda579b97c63f69e277dd7bfe063e06dab73/app/request_parsers/video_fps_test.py#L16) and the [validator](https://github.com/tiny-pilot/tinypilot/blob/4d0fdda579b97c63f69e277dd7bfe063e06dab73/app/request_parsers/validators/video_fps_test.py#L8).
- The request parsing tests all have their own definition of [`make_mock_request`](https://github.com/tiny-pilot/tinypilot/blob/09f56535282b15affb03dbd09c4adea9e360b14f/app/request_parsers/video_fps_test.py#L8), which we have 7 identical copies of in Pro.
- We are not consistent with the code style: in some cases, we just call it [`parse`](https://github.com/tiny-pilot/tinypilot/blob/4d0fdda579b97c63f69e277dd7bfe063e06dab73/app/request_parsers/video_fps.py#L6), whereas in others we call it [`parse_XYZ`](https://github.com/tiny-pilot/tinypilot/blob/4d0fdda579b97c63f69e277dd7bfe063e06dab73/app/request_parsers/hostname.py#L6).

I think it would be worth to re-evaluate the current pattern and see whether we can simplify and streamline it.

cc @jdeanwallace

jdeanwallace, 2021-08-25:
I agree with collapsing validation into the parsing step. Something @mtlynch convinced me of in the [shopify-license-check repo](https://github.com/tiny-pilot/shopify-license-check/pull/3#pullrequestreview-725580970):

> My general philosophy with parsing and validating used to be that the parser is responsible for getting data to the right types and the validator is responsible for making sure the data matches business rules. Since reading [Parse, don't validate](https://lexi-lambda.github.io/blog/2019/11/05/parse-don-t-validate/), I've come around to the idea that parsing and validating are the same operation.

mtlynch, 2021-09-03:
Yeah, my initial thinking was that the parser was responsible for converting data from one format to another (e.g., make sure fields exist in the data type we expect), whereas the validator was responsible for applying business logic (e.g. make sure that FPS is between 0 and 30).

The advantage of splitting it out this way is that it makes the "is this a valid value?" tests very clear and simple:

```python
    def test_reject_integers_out_of_bounds(self):
        self.assertFalse(video_fps.validate(0))
        self.assertFalse(video_fps.validate(31))
```

vs.

```python
    def test_reject_integers_out_of_bounds(self):
        with self.assertRaises(errors.InvalidVideoFpsError):
            video_fps.parse(make_mock_request({'videoFps': 0}))
        with self.assertRaises(errors.InvalidVideoFpsError):
            video_fps.parse(make_mock_request({'videoFps': 31}))
```

But we end up writing both tests anyway, so I agree we should just collapse all the logic into the parsers.

>We are not consistent with the code style: in some cases, we just call it parse, whereas in others we call it parse_XYZ.

My preference is to stick with `parse`. Otherwise, we end up with stutter naming. Consider:

* `request_parsers.video_fps.parse_video_fps(flask.request)` (stutter)
* `request_parsers.video_fps.parse(flask.request)` (no stutter)

But the advantage of `parse_video_fps` is that it's easily greppable.

rw4nn, 2021-12-07:
I'm starting on this, I've analyzed a bit the `parse` vs `parse_XYZ` issue.

I think it all lies in how do we want to structure the request parsers: do we want one file per "module" or one file per "thing to parse"?

For example for `video_fps` we have one file which only parses `video_fps`.

For `mass_storage` (in pro) we have one file which parses `backing_file_name`, `mount_mode`, `content_length`, `url`, each in a separate function.
(And as a side-note for `mass_storage` we can't put everything in the same `parse` method to return one object with all the data as properties, because it is called from different API methods, each one expects a different set of properties)

I would prefer the "one file per module" structure: in that case we would refactor `video_fps` and `video_jpeg_quality` into `video_settings` request parser, and keep `mass_storage` as is.

What do you think @mtlynch ?

mtlynch, 2021-12-07:
Yeah, I think it's easier to drop the `[type].parse()` convention because then we end up with a million files.

Let's make the convention that the module describes the feature they're related to (e.g., `mass_storage.py` or `video_settings.py`), and they can contain 1 or more `parse_XXX()` functions. And then the public functions are always `parse_XXX` like `parse_video_fps()` or `parse_backing_file_name()`.

One thing to keep in mind with this issue is that we want to [scope the changes narrowly](https://mtlynch.io/code-review-love/#5-narrowly-scope-changes) for each PR. So I'd like to separate the PRs that get rid of validators from the PRs that change `parse` function names.

rw4nn, 2021-12-07:
Great. Sure, I'll do separate PRs :ok_hand:

rw4nn, 2022-01-19:
This is finally done! :tada:

closed 2022-01-19 by rw4nn

jotaen, 2022-01-19:
🥳 Awesome @rw4nn, I think we’ve reached a simple and clean state now!

mtlynch, 2022-01-19:
Nice! Great work, @rw4nn!