tiny-pilot/tinypilot #785: Label pro features as 'pro'

e-minguez, 2021-09-08:
For example the 'Virtual Media' button is only for pro versions. It would be nice for non pro users to have those labeled or grayed :)

jotaen4tinypilot, 2021-10-04:
We could also consider providing a bit more info about the Pro features in the respective overlays, because people might not know what these features are about (and hence they don’t know what they actually miss). E.g., “With Virtual Media you can upload and mount disk images as USB or CD-ROM drive.” (Optionally, with an additional “Click here” link to the docs – which I’m not sure we have, though?)

jotaen4tinypilot, 2021-10-04:
Also, the menu item “Upgrade” could be “Upgrade to Pro” or something more distinct from “Update”. Otherwise it might cause confusion to see “Update” and “Upgrade” next to each other. The term “upgrading” can also be used in the context of versions (but we refer to the license here).

mtlynch, 2021-10-04:
>Also, the menu item “Upgrade” could be “Upgrade to Pro” or something more distinct from “Update”. Otherwise it might cause confusion to see “Update” and “Upgrade” next to each other. The term “upgrading” can also be used in the context of versions, but we refer to the license here.

@jotaen4tinypilot - That's a good idea! Just added: https://github.com/tiny-pilot/tinypilot/pull/800

jotaen4tinypilot, 2022-01-14:
I have some thoughts around this, so I was hoping that we can scope the task it a little. Here is a brain dump and some starting points for a discussion. Depending on what you think, we can also chat next week (or so) in person about it.

# Labels for menu items
I was thinking along the lines of:

[image]

[image]

[image]

As shown above, I’d think it’s cool to base the indicator off the brand color (the plutonium green), because the menu item is actually somewhat brand-related. But it’s not necessary, we can also do something more subtle:

[image]

This is purely stylistic, so would you have a preference? We can also try a combination, or pick one and iterate, or I can make even more suggestions.

# “Pro” feature dialog
A very minimalistic improvement would be this:
[image]

- I would include some very brief info what the feature is actually doing, just for those users who don’t know what it’s about. (Otherwise the dialog is a bit of a dead end.)
- I’m not sure what to think of the “Upgrade to Pro” button, because people also might be afraid to click it, in case they think that something might happen right away when clicking it. Something like “Learn more” would sound more open and inviting to me.
- Unrelated to this task, we could also consider providing a dedicated landing page that is meant for Community-edition users. (Some more about this below.)

Of course, we can go as far as we want with making the dialog more “fun” and engaging, both layout-wise and in regards to the information presented.

# More “Pro” dialogs?
We also have “Security” and “Wake on LAN” as Pro-exclusive menu items. Do we want to include them as well, or is it enough to only show-case “Virtual Media”?

# “Upgrade to Pro” menu item
To me, the current “Upgrade to Pro” menu item seems a bit randomly placed in between the others. I think we could consider separating it from the others, a) in order to make it stand out more (without obtruding, though), and b) to account for it functioning different from the other menu items (which are for managing the System).

[image]

There would be other ways of doing it, so the above is just a demo to get the idea across.

Similarly to what I mentioned above about the “Learn more” button, we could consider renaming the menu item, to make it more “approachable” and to spark the interest to discover what “Pro” is about.

# Landing page
This might already go beyond the scope of this task, but I’ll mention it anyway:
- I think the landing page could be more engaging, especially when it comes to demonstrating the features themselves. Currently, there is mainly text that explains the Pro features, which doesn’t make the features super tangible. But we actually have nice “live” demonstrations already on the website (e.g. [for Virtual Media](https://tinypilotkvm.com/images/virtual-storage.gif)), which we could re-use on the landing page as well.
- It might be important for users to know how complicated the upgrade process is. At least it would be important for me, and I have to admit I’m not actually aware of that myself. Can you just transition from the existing Community installation? Do you have to start over? Can I trial? Can I ever go back from Pro to Community?

mtlynch, 2022-01-18:
Cool, those are good ideas. I like the idea of dropping the "Upgrade to Pro" menu items and just including a "Pro" indicator next to Pro features.

I like this option the best:

[image]

>I’m not sure what to think of the “Upgrade to Pro” button, because people also might be afraid to click it, in case they think that something might happen right away when clicking it. Something like “Learn more” would sound more open and inviting to me.

That's a good point. I like "Learn more."

>We also have “Security” and “Wake on LAN” as Pro-exclusive menu items. Do we want to include them as well, or is it enough to only show-case “Virtual Media”?

We should do it for Wake on LAN but not Security. I think it [rubs people the wrong way](https://news.ycombinator.com/item?id=29892472) to see security as a Pro-only feature even though it's one of our most expensive features to develop and maintain.

>Landing page

Cool, good ideas! I've created a separate work item here:

https://github.com/tiny-pilot/tinypilotkvm.com/issues/556

In terms of prioritization, I don't want to begin implementation work on this yet as there are higher-priority tasks I'd like to wrap up this milestone, but it's good to just collect ideas here as they come up.

jotaen4tinypilot, 2022-01-30:
## Todos

- [ ] (a) Put “Pro” badge next to all pro-exclusive features
- [ ] (b) Add “Wake on LAN” menu item (with “Pro” badge)
- [ ] (c) Remove “Upgrade to Pro” menu item
- [ ] (d) Change button label “Upgrade to Pro” to “Learn More”

Do we also want to enhance the Pro feature dialog itself, e.g. by providing a [brief teaser on what the feature is about](https://user-images.githubusercontent.com/83721279/149531905-069867b3-4d54-4318-8283-ac3bc11edee0.png)? Currently, [the dialog looks quite generic](https://user-images.githubusercontent.com/83721279/151700836-5809a945-075c-4788-85f5-78e06d2c8a59.png) (and also a bit “un-inviting”).

jotaen4tinypilot, 2022-02-03:
@mtlynch shall we close this ticket for now, or do you want to keep it open (in case we want to work more on this)?

mtlynch, 2022-02-03:
Yeah, let's close this for now. We can revisit in the future if we want to add more detail.

closed 2022-02-03 by mtlynch